ansys/pyaedt #1852: pyaedt.modeler.Model2D.Modeler2Dd.import_3d_cad

Maschinenausleger, 2022-10-19:
### 🔍 Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### 🐞 Description of the bug

Hi, when trying to import a step file in my Maxwell Project following issue occurs: 
Traceback (most recent call last):

pyaedt error: Error in :   File "C:\Users\zj2157\Miniconda3\envs\pyMAGNET\lib\site-packages\pyaedt\modeler\Modeler.py", line 4271, in import_3d_cad
Error in :   File "C:\Users\zj2157\Miniconda3\envs\pyMAGNET\lib\site-packages\pyaedt\modeler\Modeler.py", line 4271, in import_3d_cad
pyaedt error: (-2147352567, 'Ausnahmefehler aufgetreten.', (0, None, None, None, 0, -2147024382), None) - import_3d_cad -  GENERAL OR AEDT ERROR.
(-2147352567, 'Ausnahmefehler aufgetreten.', (0, None, None, None, 0, -2147024382), None) - import_3d_cad -  GENERAL OR AEDT ERROR.
pyaedt error: Project: N36p6_rmx_s3_Topo, Design: normal (Transient, XY), [error] Script macro error: Command execution error, please see details in message window. (1:51:01  Okt 18, 2022)
pyaedt error: Arguments with values: 
Project: N36p6_rmx_s3_Topo, Design: normal (Transient, XY), [error] Script macro error: Command execution error, please see details in message window. (1:51:01  Okt 18, 2022)
Arguments with values: 
pyaedt error:     filename = C:\eti_git\magneta-topologieopt\magneta-topologieopt\Arbeitsverzeichnis\Rotorteile.step 
pyaedt error:     healing = False 
    filename = C:\eti_git\magneta-topologieopt\magneta-topologieopt\Arbeitsverzeichnis\Rotorteile.step 
    healing = False 
pyaedt error:     refresh_all_ids = False 
pyaedt error:     import_materials = False 
    refresh_all_ids = False 
    import_materials = False 
pyaedt error: Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=import_3d_cad
False

Also when I try the old way with the oEditor command: 
  File "<input>", line 1, in <module>
  File "<COMObject SetActiveEditor>", line 2, in Import
pywintypes.com_error: (-2147352567, 'Ausnahmefehler aufgetreten.', (0, None, None, None, 0, -2147024382), None)

### 📝 Steps to reproduce

from pyaedt import Maxwell2d
app = Maxwell2d()
my_modeler = app.modeler
my_modeler.import_3d_cad(filename, healing=False, refresh_all_ids=False, import_materials=False)

### 💻 Which operating system are you using?

Windows

### 🐍 Which Python version are you using?

3.8

### 📦 Installed packages

```shell
aiohttp==3.8.3
aiosignal==1.2.0
appdirs==1.4.4
async-timeout==4.0.2
attrs==22.1.0
certifi @ file:///C:/b/abs_ac29jvt43w/croot/certifi_1665076682579/work/certifi
cffi==1.15.1
charset-normalizer==2.1.1
clr-loader==0.2.4
contourpy==1.0.5
cycler==0.11.0
fonttools==4.37.4
frozenlist==1.3.1
idna==3.4
imageio==2.22.2
kiwisolver==1.4.4
matplotlib==3.6.1
multidict==6.0.2
numpy==1.23.4
packaging==21.3
pandas==1.5.0
Pillow==9.2.0
plumbum==1.8.0
psutil==5.9.2
pyaedt==0.6.21
pycparser==2.21
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.0.post1
pytz==2022.4
pyvista==0.36.1
pywin32==304
rpyc==5.0.1
scooby==0.6.0
six==1.16.0
vtk==9.2.2
wincertstore==0.2
wslink==1.8.4
yarl==1.8.1
```

gmalinve, 2022-11-02:
Hi @Maschinenausleger ,
We have tested this feature with a .step file but couldn't reproduce the error.
Our guess is whether you are running Electronics Desktop in non graphical mode and if not if you have the same issue.
Would it be possible for you to share the step file you are attempting to import please?

Maschinenausleger, 2022-11-04:
Hi @gmalinve indeed, this was a problem with my directory. Thanks.