ansys/pyaedt #1877: Bug located in stackup.add_layer, "'Edb' object has no attribute '_stackup2'"

cyasing, 2022-10-25:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I was running this example [https://aedt.docs.pyansys.com/examples/00-EDB/00_EDB_Create_VIA.html#sphx-glr-examples-00-edb-00-edb-create-via-py](url) (00_EDB_Create_VIA.py) from PyAEDT examples section, when the example could not run fully. It shows that:
 Exeption has occured: AttributeError ('Edb' object has no attribute '_stackup2').

Ideally, the entire script should have run without errors. If it helps, I have ANSYS Electronics Desktop Student 2022R2.

### Steps To Reproduce

Just created a virtual environment, installed pyaedt using pip, and run the example code.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

aiohttp==3.8.3
aiosignal==1.2.0
appdirs==1.4.4
async-timeout==4.0.2
attrs==22.1.0
cffi==1.15.1
charset-normalizer==2.1.1
clr-loader==0.2.4
contourpy==1.0.5
cycler==0.11.0
fonttools==4.38.0
frozenlist==1.3.1
idna==3.4
imageio==2.22.2
kiwisolver==1.4.4
matplotlib==3.6.1
multidict==6.0.2
numpy==1.23.4
packaging==21.3
pandas==1.5.1
Pillow==9.2.0
plumbum==1.8.0
psutil==5.9.3
pyaedt==0.6.21
pycparser==2.21
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.0.post1
pytz==2022.5
pyvista==0.36.1
pywin32==304
rpyc==5.0.1
scooby==0.6.0
six==1.16.0
vtk==9.2.2
wslink==1.9.0
yarl==1.8.1

ring630, 2022-10-27:
@cyasing I can run the example with no problem. I don't see any problem in the code neither. Can you attach the full log?

you can also try to update pyaedt to the lastest version by 
pip install -U pyaedt

cyasing, 2022-10-28:
Here's the full log (VS Code):

PS F:\Programming\Computation Projects\pyaedt_scripts> .venv\Scripts\Activate.ps1                                                         (.venv) PS F:\Programming\Computation Projects\pyaedt_scripts>  f:; cd 'f:\Programming\Computation Projects\pyaedt_scripts'; & 'f:\Programming\Computation Projects\pyaedt_scripts\.venv\Scripts\python.exe' 'c:\Users\Knight\.vscode\extensions\ms-python.python-2022.16.1\pythonFiles\lib\python\debugpy\adapter/../..\debugpy\launcher' '55715' '--' 'f:\Programming\Computation Projects\pyaedt_scripts\00_EDB_Create_VIA.py' 
C:\Users\Knight\AppData\Local\Temp\pyaedt_prj_9GG\pcb_0ZGST3.aedb
pyaedt info: Logger started on C:\Users\Knight\AppData\Local\Temp\pyaedt_prj_9GG\pyaedt20221028_085228.log
pyaedt info: Logger is initialized in EDB.
pyaedt error: Error in :   File "f:\Programming\Computation Projects\pyaedt_scripts\.venv\lib\site-packages\pyaedt\edb.py", line 297, in _init_dlls
Error in :   File "f:\Programming\Computation Projects\pyaedt_scripts\.venv\lib\site-packages\pyaedt\edb.py", line 297, in _init_dlls     
pyaedt error: Could not load file or assembly 'Ansys.Ansoft.Edb' or one of its dependencies. The system cannot find the file specified.
File name: 'Ansys.Ansoft.Edb'
   at System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence
Could not load file or assembly 'Ansys.Ansoft.Edb' or one of its dependencies. The system cannot find the file specified.
File name: 'Ansys.Ansoft.Edb'
   at System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence
pyaedt error: assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyNam
assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyNam
pyaedt error: e(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.Ass
e(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.Ass
pyaedt error: embly.Load(AssemblyName assemblyRef)
   at Python.Runtime.AssemblyManager.LoadAssembly(AssemblyName name)
   at Python.Runtime.CLRModule.AddReference(String name)

WRN: Assembly binding logging is turned OFF.
To enable assembly bind failure logg
Arguments with values:
pyaedt error:     init_dlls = None
    init_dlls = None
pyaedt error: Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=create_edb
Check Online documentation on: https://aedtdocs.pyansys.com/search.html?q=create_edb
pyaedt info: EDB C:\Users\Knight\AppData\Local\Temp\pyaedt_prj_9GG\pcb_0ZGST3.aedb was created correctly.
pyaedt info: Failed to initialize DLLs.

[image]

ring630, 2022-10-28:
@cyasing the log says init_dll failed. It looks like it is another problem. It is difficult to tell the cause without knowing details of your venv. I will suggest to start over again by create a new venv and install pyaedt. 

In the new venv, please let me know if you can run this example without any failure. 
[pyaedt](https://github.com/pyansys/pyaedt)/[examples](https://github.com/pyansys/pyaedt/tree/main/examples)/[00-EDB](https://github.com/pyansys/pyaedt/tree/main/examples/00-EDB)/08_CPWG.py

skandak-ansys, 2022-10-30:
@cyasing
In example 00-edb-create-via-py, when an `Edb` object is created, it has a `student_version` parameter whose default value is `False`. Since you only have the student version of AEDT, the script is unable to create an `Edb` object. The solution is -  add `student_version=True` when creating the `Edb` object and it should now work fine.

ring630, 2022-10-31:
@skandak-ansys thanks for pointing out the cause. @cyasing could you try @skandak-ansys 's recommendation?

cyasing, 2022-10-31:
@ring630, @skandak-ansys, still the same thing, unfortunately:

[image]

Log:
(.vaedt) PS F:\Programming\Computation Projects\aedt_scipts> python 00_EDB_Create_VIA.py 
C:\Users\Knight\AppData\Local\Temp\pyaedt_prj_05R\pcb_AL0356.aedb
pyaedt info: Logger started on C:\Users\Knight\AppData\Local\Temp\pyaedt_prj_05R\pyaedt20221031_182556.log
pyaedt info: Logger is initialized in EDB.
pyaedt error: Error in :   File "F:\Programming\Computation Projects\aedt_scipts\.vaedt\lib\site-packages\pyaedt\edb.py", line 297, in _init_dlls
Error in :   File "F:\Programming\Computation Projects\aedt_scipts\.vaedt\lib\site-packages\pyaedt\edb.py", line 297, in _init_dlls
pyaedt error: Could not load file or assembly 'Ansys.Ansoft.Edb' or one of its dependencies. The system cannot find the file specified.
File name: 'Ansys.Ansoft.Edb'
   at System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence
Could not load file or assembly 'Ansys.Ansoft.Edb' or one of its dependencies. The system cannot find the file specified.
File name: 'Ansys.Ansoft.Edb'
   at System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence
pyaedt error: assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyNam
assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyNam
pyaedt error: e(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.Ass
e(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.Ass
pyaedt error: embly.Load(AssemblyName assemblyRef)
   at Python.Runtime.AssemblyManager.LoadAssembly(AssemblyName name)
   at Python.Runtime.CLRModule.AddReference(String name)

WRN: Assembly binding logging is turned OFF.
To enable assembly bind failure logg
embly.Load(AssemblyName assemblyRef)
   at Python.Runtime.AssemblyManager.LoadAssembly(AssemblyName name)
   at Python.Runtime.CLRModule.AddReference(String name)

WRN: Assembly binding logging is turned OFF.
To enable assembly bind failure logg
pyaedt error: ing, set the registry value [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD) to 1.
Note: There is some performance penalty associated with assembly bind failure logging.
To turn this feature off, remove the registry value [HKLM\Software\Microsoft
ing, set the registry value [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD) to 1.
Note: There is some performance penalty associated with assembly bind failure logging.
To turn this feature off, remove the registry value [HKLM\Software\Microsoft
pyaedt error: \Fusion!EnableLog].
  _init_dlls -  GENERAL OR AEDT ERROR.
Arguments with values:
pyaedt error:     init_dlls = None
    init_dlls = None
pyaedt error: Check Online documentation on: https://aedt.docs.pyansys.com/search.html?q=create_edb
Check Online documentation on: https://aedt.docs.pyansys.com/search.html?q=create_edb
pyaedt info: EDB C:\Users\Knight\AppData\Local\Temp\pyaedt_prj_05R\pcb_AL0356.aedb was created correctly.
pyaedt info: Failed to initialize DLLs.
Traceback (most recent call last):
  File "F:\Programming\Computation Projects\aedt_scipts\00_EDB_Create_VIA.py", line 29, in <module>
    edb.stackup.add_layer("GND")
  File "F:\Programming\Computation Projects\aedt_scipts\.vaedt\lib\site-packages\pyaedt\edb.py", line 636, in stackup
    if not self._stackup2 and self.builder:
**AttributeError: 'Edb' object has no attribute '_stackup2'. Did you mean: '_stackup'?**
(.vaedt) PS F:\Programming\Computation Projects\aedt_scipts>

cyasing, 2022-10-31:
@ring630 Yes, you can see it in the screenshot. A new venv, named vaedt, in a new folder aedt_scripts.

ring630, 2022-11-01:
@svandenb-dev can you help to have a look at this issue?

ring630, 2022-12-01:
@cyasing it could be related to a bug which is fixed now. Can you try again with the latest release?

closed 2022-12-15 by svandenb-dev