ansys/pyaedt #1896: Hiding Variable is slow

adamgregory81, 2022-10-28:
I noticed that hiding variables in HFSS through pyaedt is very slow.  Tried 2 techniques

1:
hfss.hidden_variable(var_name, True)

2:
hfss.variable_manager.variables[var_name].hidden = True

Both techniques seem comparable.  It was taking on the order of 5 - 10 seconds to hide 80 variables.
In the HFSS GUI, I could hide all the variables, and it appeared to happen instantaneously.

Samuelopez-ansys, 2022-11-11:
Hi @adamgregory81 ,

I have implemented a new option in the "hidden_variable" method. Now it allows a list of variables like:
aedtapp.hidden_variable(["my_oo_variable", "$dim", "my_hidden_variable"]).

This will hide all these variables, and it will be as fast as the UI.

It will available in the next pyaedt release. 

Thanks

MaxJPRey, 2022-11-12:
Well done @Samuelopez-ansys . This will change the performance of the scripts execution for the best.

closed 2022-11-14 by Samuelopez-ansys

adamgregory81, 2022-11-14:
Thanks for the update and the fix

adamgregory81, 2023-04-24:
Using the latest version of Pyaedt (v0.6.71), I found that hiding variables is still slow.  I noticed that the code used in the November 14th 2022 commit (#1977) in Design.py no longer appeared to be there.  Referring to lines 1856 - 1878 in that commit.  As a test, I implemented that code myself, and it was still slow.  Debugging a bit, I discovered the culprit was fetching these lists inside of the loop:
self.variable_manager.design_variable_names
self.variable_manager.project_variable_names

When I changed to fetching them a single time outside of the loop, I am able to hide 100+ variables as fast as the HFSS GUI.