ansys/pyaedt #1911: error message occur when open aedt file with pyaedt version "0.6.22"

alien008, 2022-11-01:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

i just install the newest version of pyaedt and start to run a simple code , the code could run but output some errors then proceed to run until finished. but the error message is wired.
here is  my code: "
    with Hfss(projectname="amf_GSG_SW_lumped_1000um.aedt", designname="HFSSDesign1", new_desktop_session=True,
              close_on_exit=True, student_version=False, specified_version="2022.1") as hfss:
        hfss.autosave_disable()
        objects = hfss.modeler.primitives.object_names
        print(objects)"
which is simple and no error occur before until i install the new pyaedt version
here is the output message:
      **Exception in thread pyaedt info: Logger Started
      pyaedt info: Launching PyAEDT outside AEDT with CPython and PythonNET.
      pyaedt info: AEDT installation Path C:\Program Files\AnsysEM\v221\Win64.
     ** _Thread-1:
      Traceback (most recent call last):
        File "C:\Users\zgl\anaconda3\lib\threading.py", line 932, in _bootstrap_inner
          self.run()
        File "C:\Users\zgl\anaconda3\lib\threading.py", line 870, in run
          self._target(*self._args, **self._kwargs)
        File "C:\Users\zgl\anaconda3\lib\site-packages\pyaedt\application\Design.py", line 163, in load_aedt_thread
          settings._project_properties[path] = load_entire_aedt_file(path)
        File "C:\Users\zgl\anaconda3\lib\site-packages\pyaedt\generic\LoadAEDTFile.py", line 26, in load_entire_aedt_file
          return _load_entire_aedt_file(os.path.normpath(filename))
        File "C:\Users\zgl\anaconda3\lib\site-packages\pyaedt\generic\LoadAEDTFile.py", line 377, in _load_entire_aedt_file
          _read_aedt_file(filename)
        File "C:\Users\zgl\anaconda3\lib\site-packages\pyaedt\generic\LoadAEDTFile.py", line 347, in _read_aedt_file
          with open_file(filename, "rb") as aedt_fh:
      AttributeError: __enter__**
      pyaedt info: Launching AEDT with module PythonNET.
      pyaedt info: Ansoft.ElectronicsDesktop.2022.1 Started with process ID 4864.
      pyaedt info: Logger file C:\Users\zgl\Documents\Ansoft\pyaedt20221101_103621.log in use.
      pyaedt info: pyaedt v0.6.22
      pyaedt info: Python version 3.8.12 (default, Oct 12 2021, 03:01:40) [MSC v.1916 64 bit (AMD64)]
      pyaedt info: Project amf_GSG_SW_lumped_1000um has been opened.
      pyaedt info: Aedt Objects initialized
      pyaedt info: Variable Manager initialized
      pyaedt info: Design Loaded
      pyaedt info: Successfully loaded project materials !
      pyaedt info: Materials Loaded
      pyaedt info: aedt file load time 0.031244993209838867
      ['sub', 'box', 'cladding', 'air']

### Steps To Reproduce

just run the code :
with Hfss(projectname="amf_GSG_SW_lumped_1000um.aedt", designname="HFSSDesign1", new_desktop_session=True,
              close_on_exit=True, student_version=False, specified_version="2022.1") as hfss:
        hfss.autosave_disable()
        objects = hfss.modeler.primitives.object_names
        print(objects)"
the aedt file could be substitute by your own files

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

aiohttp==3.8.3
aiosignal==1.2.0
[... 274 lines cut ...]
Unidecode @ file:///tmp/build/80754af9/unidecode_1614712377438/work
urllib3==1.26.7
vtk==9.1.0
watchdog @ file:///C:/ci/watchdog_1638349261759/work
wcwidth @ file:///Users/ktietz/demo/mc3/conda-bld/wcwidth_1629357192024/work
webencodings==0.5.1
Werkzeug @ file:///tmp/build/80754af9/werkzeug_1635505089296/work
whichcraft @ file:///tmp/build/80754af9/whichcraft_1617751293875/work
widgetsnbextension==3.5.1
win-inet-pton @ file:///C:/ci/win_inet_pton_1605306167264/work
win-unicode-console==0.5
wincertstore==0.2
wrapt==1.12.1
wslink==1.3.1
xlrd @ file:///tmp/build/80754af9/xlrd_1608072521494/work
XlsxWriter @ file:///tmp/build/80754af9/xlsxwriter_1636633762820/work
xlwings==0.24.9
xlwt==1.3.0
xmltodict @ file:///Users/ktietz/demo/mc3/conda-bld/xmltodict_1629301980723/work
yapf @ file:///tmp/build/80754af9/yapf_1615749224965/work
yarl==1.7.2
zict==2.0.0
zipp @ file:///opt/conda/conda-bld/zipp_1641824620731/work
zope.event==4.5.0
zope.interface @ file:///C:/ci/zope.interface_1625036260333/work

gmalinve, 2022-11-02:
Hi @alien008 ,
We have tested it but couldn't reproduce your bug.
This is a pyaedt core feature and being extensively used it has seldom given back any exception. 
Maybe could you share your aedt file so we can properly test and see whether it's a pyaedt or a file issue?

alien008, 2022-11-14:
Hi,gmalinve,
Sorry for replying so late, i tested one again, even reinstalled anaconda. but there is still the message.
the attachment  contain the .py file and aedt file.
Thank you for your help!
[amf_GSG_SW_lumped_1000um.zip](https://github.com/pyansys/pyaedt/files/9998874/amf_GSG_SW_lumped_1000um.zip)

alien008, 2022-11-14:
Yes, i have updated to version 0.6.26, but the error message still occurs, Thank you!

alien008, 2022-11-14:
[image]
 this is the pycharm output,thanks!

gmalinve, 2022-11-14:
Hi @alien008 ,

I tested your file with both versions 22.1 and 22.2 and didn't get any issue.
Have you checked that you don't have the .lock file in your folder? because this will prevent the opening of the aedt file.

alien008, 2022-11-16:
Hi,gmalinve,
well, i think it's not because of the .lock file. I know that the .lock file will cause the aedt file to fail to open, but there is no .lock file under my folder, and the aedt file can be opened by python, and pycharm will output such an error, But the aedt file will execute normally. If I run the same file under pyaedt 0.4.23, no error will be reported. I do not know why either. Thank you!

alien008, 2022-11-16:
[image]

gmalinve, 2022-11-22:
Hi @alien008 , yes at the moment we are investigating/fixing this case. A PR has been open to tackle it.

MaxJPRey, 2022-11-23:
Hi @alien008
This should be fixed within the last release 0.6.28 published yesterday.
Let us know if you still experiment any issues.
Thanks for your feedback and all the details you provided.

closed 2022-12-02 by maxcapodi78