ansys/pyaedt #1912: Enabling the monitor point export from the Icepak design settings

TejasJe, 2022-11-01:
### Description of the feature

A new key needs to be added to the "edit_design_settings" method in the Icepak class to enable the monitor point data export after the completion of the simulation. 

The export directory override also needs to be created defaulting to the current working directory.

The option shown in the image should available for enablement.

[image]

### Steps for implementing the feature

Additional method arguments and the dictionary keys to be added in the method "edit_design_settings". 

"ExportOnSimulationComplete:=", for activating the radio button 
   "ExportDirectory:=", for export directory override

Please refer the following images

[image]
[image]

Samuelopez-ansys, 2022-11-11:
Hi @TejasJe ,

This issue was solved, right? 

Please if it is, close the Issue.

Regards

TejasJe, 2022-11-11:
Hello @Samuelopez-ansys,

Yes, this issue is resolved. I will close this issue. 

Thanks

closed 2022-11-11 by TejasJe