ansys/pyaedt #1937: Not possible to instantiate hfss

sara-tell, 2022-11-04:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

when instantiating hfss with pyaedt.hfss, the pyaedt version 0.6.23 is printed out in the Message Manager in AEDT gui maybe 50 times or so. A lot of errors with log handler and messages occur in python and finishes with: RecursionError: maximum recursion depth exceeded while calling a Python object

### Steps To Reproduce

run

pyaedt.Hfss(
    specified_version="22.2",
    solution_type="SBR+",
    new_desktop_session=True,
    projectname="projectname",
    close_on_exit=True,
)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

aiohttp==3.8.3
aiosignal==1.2.0
appdirs==1.4.4
async-timeout==4.0.2
attrs==22.1.0
cffi==1.15.1
charset-normalizer==2.1.1
clr-loader==0.2.3
contourpy==1.0.5
cycler==0.11.0
fonttools==4.37.3
frozenlist==1.3.1
idna==3.4
imageio==2.22.0
kiwisolver==1.4.4
matplotlib==3.6.0
multidict==6.0.2
numpy==1.23.3
open-simulation-interface==3.4.0
packaging==21.3
pandas==1.5.0
Pillow==9.2.0
plumbum==1.7.2
protobuf==4.21.6
psutil==5.9.2
pyaedt==0.6.23
pycparser==2.21
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.0.post1
pytz==2022.2.1
pyvista==0.36.1
pywin32==304
rpyc==5.0.1
scooby==0.6.0
six==1.16.0
vtk==9.1.0
wslink==1.8.2
yarl==1.8.1

MaxJPRey, 2022-11-04:
@sara-tell
Thanks for your feedback.
It should be fixed in the new release that we are about to publish today (0.6.24).
We yanked 0.6.23.
Sorry for the inconvenience.
Please let us know if you still experiment any limitations with the next release.
Thanks.

sara-tell, 2022-11-11:
Hi, yes it is fixed.

closed 2022-11-11 by Samuelopez-ansys

MaxJPRey, 2022-11-16:
Thanks again @sara-tell for the feedback.