ansys/pyaedt #1995: sweep_along_path doesn't work with Parasolid kernel in 2022R2

IreneWoyna, 2022-11-16:
it works with 2022R2 ACIS and 2023R1 Parasolid, but 2022R2 throws 2 errors:

 [error] Parasolid Error: Sweep failed function PK_BODY_make_swept_body_2 type PK_BODY_sweep_check_failure_c.  (4:22:06 pm  Nov 16, 2022)
 [error] Body could not be created for part because of invalid parameters to SweepAlongPath operation. (4:22:06 pm  Nov 16, 2022)

MaxJPRey, 2022-11-16:
Hi @IreneWoyna,
As discussed internally, because this limitation has been fixed with the new release to come (2023R1), can we close this issue?

MaxJPRey, 2022-11-17:
No problem. Thanks @IreneWoyna .

closed 2022-11-17 by MaxJPRey

Gettikh, 2024-11-14:
Hello colleagues,

if one creates a polyline with height and depth (rectangle shape) with help of my PyAEDT script, one get the following error message:
 _Parasolid Error: Sweep failed function PK_BODY_make_swept_body_2typePK_body_sweep_nong1_failure_c ._
I see, the same issues was created in 2022R2.
Is the problem already known or the issues was fixed?

Thank you,
Andrey