ansys/pyaedt #1996: edit_y_axis_scaling not working

GiadaVent, 2022-11-17:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The function edit_y_axis_scaling is not working properly in version 0.6.26. I guess it is due to the double parenthesis:
        if units:
            props.append((["NAME:Auto Units", "Value:=", False]))
            props.append(["NAME:Units", "Value:=", units])
        return self._change_property("Scaling", "Axis" + axis_name, props)

### Steps To Reproduce

use the function trying to change the axis unit

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

(pyCharm_Git) PS C:\Workspace\pymotor-dev> python -m pip freeze
aiohttp==3.8.3
aiosignal==1.2.0            
anyio==3.6.1                
appdirs==1.4.4              
argon2-cffi==21.3.0         
argon2-cffi-bindings==21.2.0
asttokens==2.0.8            
async-timeout==4.0.2
attrs==22.1.0
Babel==2.10.3
backcall==0.2.0
beautifulsoup4==4.11.1
bleach==5.0.1
certifi==2022.9.24
cffi==1.15.1
charset-normalizer==2.1.1
clr-loader==0.2.3
colorama==0.4.5
contourpy==1.0.5
cycler==0.11.0
debugpy==1.6.3
decorator==5.1.1
defusedxml==0.7.1
entrypoints==0.4
executing==1.1.0
fastjsonschema==2.16.2
fonttools==4.37.3
frozenlist==1.3.1
idna==3.4
imageio==2.22.0
importlib-metadata==4.12.0
importlib-resources==5.9.0
ipykernel==6.16.0
ipython==8.5.0
ipython-genutils==0.2.0
jedi==0.18.1
Jinja2==3.1.2
json5==0.9.10
jsonschema==4.16.0
jupyter-core==4.11.1
jupyter-server==1.19.1
jupyter_client==7.3.5
jupyterlab==3.4.7
jupyterlab-pygments==0.2.2
jupyterlab_server==2.15.2
kiwisolver==1.4.4
lxml==4.9.1
MarkupSafe==2.1.1
matplotlib==3.6.0
matplotlib-inline==0.1.6
mistune==2.0.4
multidict==6.0.2
nbclassic==0.4.3
nbclient==0.6.8
nbconvert==7.0.0
nbformat==5.6.1
nest-asyncio==1.5.5
notebook==6.4.12
notebook-shim==0.1.0
numpy==1.23.3
nums-from-string==0.1.2
packaging==21.3
pandas==1.5.0
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.2.0
pkgutil_resolve_name==1.3.10
plumbum==1.7.2
prometheus-client==0.14.1
prompt-toolkit==3.0.31
psutil==5.9.2
pure-eval==0.2.2
pyaedt==0.6.26
pycparser==2.21
Pygments==2.13.0
pyparsing==3.0.9
pyrsistent==0.18.1
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.2.1
pyvista==0.36.1
pywin32==304
pywinpty==2.0.8
pyzmq==24.0.1
requests==2.28.1
rpyc==5.0.1
scipy==1.9.1
scooby==0.6.0
Send2Trash==1.8.0
six==1.16.0
sniffio==1.3.0
soupsieve==2.3.2.post1
stack-data==0.5.1
terminado==0.16.0
tinycss2==1.1.1
tomli==2.0.1
tornado==6.2
traitlets==5.4.0
urllib3==1.26.12
vtk==9.1.0
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.4.1
wslink==1.8.2
yarl==1.8.1
zipp==3.8.1

gmalinve, 2022-11-17:
Hi @GiadaVent ,
I could not replicate your issue. I tried to edit the y axis scaling in this way and it worked:
`report = hfss.post.create_report("dB(S(1,1))")
report.edit_y_axis_scaling(
            linear_scaling=False, min_scale="-50", max_scale="10", minor_tick_divs=10, min_spacing="5"
        )`
could you please share the code snippet you are trying to run?
Thanks!

GiadaVent, 2022-11-17:
Hello Giulia,
I am trying to change the axis unit. Find an example below:
m2D.post.plots[n].edit_y_axis_scaling(axis_name='Y1', units='W')
Cheers

gmalinve, 2022-11-17:
Hi @GiadaVent,
could you please post the type of error and the message you get?
do you have an axis named Y1 in your plot? and what report type are you trying to modify?
Thanks!

closed 2022-11-21 by gmalinve