ansys/pyaedt #2003: Converting a program that output optimetrics scan data to a single result from Setup1

jpotter0, 2022-11-18:
I am far from an expert in Python programming. Access to HFSS data to print it out is beyond me. This is a short section of code that worked fine when I was printing out data from an optimetrics scan. I have been able to figure out how to run a single instance of Setup1, but I don't know how to access the real part of the eigenmode frequency. The issue is with the print line. However, I also don't understan the lines about variations.

[jpaw3.txt](https://github.com/pyansys/pyaedt/files/10037715/jpaw3.txt)

jpotter0, 2022-11-18:
Here are the messages I get on HFSS. I can't figure out where the ghost of OptimizationSetup2 comes from. For some reason I am unable to gett the data.
[HFSS messages.pdf](https://github.com/pyansys/pyaedt/files/10038334/HFSS.messages.pdf)

Samuelopez-ansys, 2022-11-18:
@jpotter0 If you need to access the real part using the method get_solution_data, you just need:

data.data_real()

jpotter0, 2022-11-18:
Samuel,

There's so much to learn, but I'm making progress. Not too bad for an 81 year old. Proof that old dogs can learn new tricks!

This represents a major step toward my goal. I now need to read in some parameters from a  CVS file. I already figured that out a couple of years ago. Then I need to create cells from my data and ultimately combine them to make an accelerator model. I've already done that the old way for a different type of accelerator.

Sometimes getting pointed in the right direction is all that I need. The debugger is a great teaching tool as long as I am not too far of the path.

This is my idea of fun.

Best regards,

Jim

closed 2022-11-19 by jpotter0