ansys/pyaedt #2012: Lots of modeler questions

jpotter0, 2022-11-21:
I am working on a program to create a resonant cavity using the modeler. I am overwhelmed with information and struggling with finding my way around the documentation. I don't see a comprehensive list of functions where I can find what I need. Here is a list of my questions. Hopefully, I can get some guidance so I don't spend hours looking for the information. I know how to do all this stuff in int the EDT and before aedt I created a Python code to create a complicated arrangement of resonant cells using the script record feature to find out what to do. With pyaedt I am flying under the hood and trying not to hit mountains.

First of all I have figured out how to create a polyline. I also figured out how to create cylinders. The items are sort of listed in the order I need to deal with them.

1. How to I cover a surface and sweep it along a vector?
2. How do I do Boolean operations on the solids I've created? Unite? Subtract?

Once I've created the solid I need to set up boundary conditions. The model has a lot of faces. I need to be able to identify them.

3. How do I reference faces to set boundary conditions?

I need to set two faces in the XY plane at z=0.0 and at z="Length" to symmetrical H? I need to set another face in the YZ plane at x=0.0 to symmetrical H. After that I need to set all faces except the 5 as conducting surfaces, It is important to be able to identify the features in a way that doesn't change with changes to the model. Previous I used the face numbers, but geometry changes kept getting some of the face numbers changed. Every time that happened I had to identify the faces by number all over again. I would think it should be pretty simple since all the non-conducting faces are in symmetry planes,

I will first create individual a cells find and adjust their resonant frequency. I have developed a Python routine that uses a simplified Newton's method approach. I can find the frequency at two values of my parameter to obtain the sensitivity to the parameter. I was going to make a loop to find convergence, but one iteration seems to be adequate, The sensitivity varies a bit with the geometry, but is pretty linear over small range. I have configured the mesh to converge to within 0.03% in 4 iterations, ~100 kHz out of 352.5 MHz. It takes about 2.5 minutes. That means less than 8 minutes per cell. That's faster and more accurate than I can do with Optimetrics, which is important since I have several 100 cells to do. It take about an hour to do what requires 6 to 8 hours in Optimetrics with the quasi Newton method. The problem is that the quasi Newton is looking for the minimum of a quadratic and is susceptible to mesh noise at the level I'm looking for. This would not be an issue if the small changes merely stretched the mesh rather than recreating a mesh for each iteration. I've been bugging ANSYS for years to add this feature. I did it years ago when I had to create meshes by hand for a 2D resonator finite element code running on a 12 MHz 8086 system with 1 MB of RAM (~1985).

4. I need to create field maps for a region near the axis. I know how to do this in the EDT, but, so far, I've seen no reference to this in the documentation.

I'll probably come up with more questions as I move along and I may stumble upon some answers myself as I did for the polylines and cylinders, but I'd appreciate any advice that points me in the right direction. I find the existing documentation comprehensive but overwhelming and there are no HFSS examples for resonant cavities. The examples that do exist seem to all use imported models and have no modeler information. I have been using HFSS since 1999 with version 5, which, if I remember correctly, cost about $5k and ran on a $8k computer. The computer has gotten cheaper, but not the software. I appreciate how much the software has improved over the years and know that there's no way I could write an equivalent program in a hundred lifetimes!

Thanks,

Jim (JP Accelerator Works, Inc.)

jpotter0, 2022-11-21:
Some progress since I last posted. _ found the list of primitives under Modeler and Primitives\paedt.modeler,Modeler3D.Modeler.3D. I was able to cover the polyline, Then I found I could specify it when I created it as a closed polyline,

After bit of trial and error, I am apparently able to Sweep the surface.

After I create the other components and create a single vacuum volume I'll need to come to grips with setting the boundary conditions, running the analysis, and mapping the fields.

Jim

jpotter0, 2022-11-21:
Lots of progress now. Some things even make sense. :-) A problem has cropped up. I need to make a new coordinate system to split off a cylinder that extends beyond a symmetry plane that is parallel to the XY plane bit not at Z=0.

    cs_name="cs"+str(cellnum)
    cs.append(aedtapp.modeler.create_coordinate_system(origin=[0,0,str(zend)],reference="Global",name=cs_name))
    aedtapp.modeler.set_working_coordinate_system(cs_name)

HFSS sees the second command but says there is no coordinate system named "cs1". If I get past this my model wil be complete except for selecting all but the symmetry plane to be conducting surface.

Jim

jpotter0, 2022-11-21:
Found the problem. it should say reference_cs="Global"

fwlfHS, 2022-11-22:
Hi Jim, 
you get the all faces of an object by "object.faces" or a specific face by "object.bottom_face_z" , were you can use bottom and top in every direction (x,y,z)
To a assign a boundary just use desired function e.g. "hfss.assign_radiation_boundary_to_faces(object.faces)

jpotter0, 2022-11-22:
Thanks! That sounds like exactly what I am looking for.

My project can now generate N resonant cavities (cells for an accelerator) and combine them into a tank, but I need a different version that can use most the same code. I have made simple changes to the code to generate all of the cells at the same location, but I need each cell to be in a different design in the same project.

Also, the program keeps incrementing a generic project name. I'd like to be able to specify the design name. I know this must be simple, but I am overwhelmed with information.

fwlfHS, 2022-11-22:
you define  project and design during initialization: hfss = Hfss("HfssProject","HfssDesign1") https://aedt.docs.pyansys.com/API/_autosummary/pyaedt.Hfss.html

jpotter0, 2022-11-22:
OK. Presumably, I can create a number of designs hfss1, hfss2, etc at intialization, how do I specify which one inside the loop that creates the cells? I can create the name as design = "hfss"+str(cellnum). I'm doing that for all my parameter and object names inside the loop. I learned that a couple of years ago when a created a similar Python program for a different accelerator programing with HFSS scripts the hard way. pyaedt is definitely more straightforward.

I'm adding your earlier suggestion to my code, but I got stopped at the assign part. Where do I get information on the boundary functions? what I need in every case is symmetry, pecfectH. Also, I need conductivity for all faces except those boundaries.

I really appreciate your help. Thanks!

fwlfHS, 2022-11-22:
I'd recommend to search the API reference (https://aedt.docs.pyansys.com/API/), since you know all the functions in the GUI. 
If you search for "symmetry" you get the hfss.assign_symmetry() function pretty fast. There is a hfss.assign_perfecth_to_sheets() as well. The finite conductivity boundary is hidden in hfss.assign_coating()

jpotter0, 2022-11-22:
OK, thanks again! I'm an amateur at this. I have been programming since 1964 (FORTRAN) for my own purposes. My approach has always been to learn what I need for my own applications. I currently use LabVIEW and ANSI C, and I've been learning Python and now pyaedt. I'm living proof that you can teach and old dog new tricks, and I'm enjoying every minute!

jpotter0, 2022-11-23:
I am trying to implement your suggestion:
"you get the all faces of an object by "object.faces" or a specific face by "object.bottom_face_z" , were you can use bottom and top in every direction (x,y,z)"

From looking at the API it appears to me that

tp_face__z = pyaedt.modeler.objects.top_face_z

when I used 'object' I got a message saying there was no such item in modeler.
This morning a typo where I used 'objects' cleared away the error message. I was hopeful that would work, but when I ran the code it crashed with a message: AttributeError" 'dict' object has no attribute 'top_face_z'

I know there has to be some object that works, and I know there is a primitive top_face_z. At the moment the mystery is what is 'dict'?

fwlfHS, 2022-11-23:
object should be the python name of the 3D object you created. So if you create box like in the example below replace object with "box_object" which is the python name of your HFSS box.

```
from pyaedt import hfss
hfss = Hfss()
origin = [0,0,0]
dimensions = [10,5,20]
box_object = hfss.modeler.primivites.create_box(origin, dimensions, name="mybox", matname="copper")
```

dict is a dicionary which is a python data type (https://realpython.com/python-dicts/)

jpotter0, 2022-11-23:
OK, that's useful information. Here's where my object gets its name:

    cell_points.append([[bhw, -btm[cellnum], zb[cellnum]], [bhw, -Lstem - Ltpr, zb[cellnum]], [thw,-Lstem, zb[cellnum]],
                        [0.0,-Lstem, zb[cellnum]],[0.0,Lstem, zb[cellnum]], [thw,Lstem, zb[cellnum]],[bhw, Lstem+Ltpr,
                        zb[cellnum]],[bhw, btm[cellnum], zb[cellnum]],[rcell[cellnum],0, zb[cellnum]],[bhw,
                        -btm[cellnum], zb[cellnum]]])
    cell_name="cell"+str(cellnum)
    Cell.append(aedtapp.modeler.create_polyline(cell_points[cellnum], segment_type=["Line", "Line","Line","Line",
                        "Line","Line","Line","Arc"],cover_surface= True,name=cell_name))

    Cell_outer.append(aedtapp.modeler.sweep_along_vector(cell_name,[0.0,0.0,Len_cell]))

Cell is a list of objects.

Here's where I try to use the name.

    obj_name = Cell[cellnum]
    tp_face_z = aedtapp.modeler.obj_name.top_face_z
    btm_face_z = aedtapp.modeler.obj_name.bottom_face_z
    btm_face_x = aedtapp.modeler.obj_name.bottom_face_x
    aedtapp.assign_symmetry([btm_face_z,tp_face_z,btm_face_x],symmetry_name="Perfect H",is_perfect_e=False)

Here is the error message I get:

  File "C:\Jims Files\PY Scripts\ihdtl3.py", line 213, in <module>
    tp_face_z = aedtapp.modeler.obj_name.top_face_z
AttributeError: 'Modeler3D' object has no attribute 'obj_name'

Inside the EDT the first object Cell[1] is called cell1. Right now I am just creating one cell so I can set its boundary conditions and set its radius to get it on frequency.

Here is a screen snip of the HFSS Editor window;
[image]

Thanks,
Jim

What am I doing wrong?

jpotter0, 2022-11-23:
That was the fix, thanks! I don't think I could have figured that out by myself.

In the earliest days of my programming career I was writing some code that ran on "Stretch", which was at that time the world's fastest computer. Well, it had a twin at Livermore. I kept a notebook entitled "Things They Never Tell You Until You Ask."

Now I am on to an error that doesn't crash pyCharm, With the previous error I had to recover by using Task Manager to End Task for pyCharm and HFSS. Nothing I tried in pyCharm would eliminate the message that my program was still running and allow me to recover gracefully.

Now I have a more benign error: pyaedt ERROR: Error in boundary creation for Symmetry Perfect H.

It's like peeling an onion. You remove one mistake and there's another underneath. :-)

Jim

jpotter0, 2022-11-24:
I solved the problem with setting boundary conditions Here is my code:
    obj_name = Cell[cellnum]

    faces = obj_name.faces_on_bounding_box

    sym_name1 = "sym1-cell"+str(cellnum)
    sym_name2 = "sym2-cell"+str(cellnum)
    sym_name3 = "sym3-cell"+str(cellnum)
    face_id1 = faces[7]
    face_id2 = faces[8]
    face_id3 = faces[2]
    aedtapp.assign_symmetry([face_id1],symmetry_name=sym_name1,is_perfect_e=False)
    aedtapp.assign_symmetry([face_id2],symmetry_name=sym_name2,is_perfect_e=False)
    aedtapp.assign_symmetry([face_id3],symmetry_name=sym_name3,is_perfect_e=False)

There is a problem with using top_face_z, for example, if there is more than one face on that side. It picks one. It might even pick a curved surface and ignore the planar faces that are on the bounding box surface.

The faces on bounding box gets a list of all the faces. I recorded the faces that I wanted to use and set the boundary condition for each. when I iterate on a parameter the face name changes, but the position in the list does not.

Now I am trying to create each cell in a different design. The below code creates the desired design, but AEDT remains in the first design.

for num in range(ncells):
    cellnum = num + 1
    design_name = "ihdtl"+str(cellnum)
    pyaedt.Hfss(designname = design_name, solution_type = "Eigenmode")

    Hfss.set_active_design(design_name)

The pyaedt.Hfss line creates a new design on every iteration, but does not select it.

the set_active_design generates an error:

pyaedt ERROR: set_active_design() missing 1 required positional argument: 'name' - set_active_design -  TYPE ERROR.
pyaedt ERROR: Arguments with values: 
pyaedt ERROR:     name = None 

I can't figure out what to do. The API reference shows the lines below, with no hint about what the requirements are for the name. Apparently, it doesn't like text, but nothing I've tried fixes it.

Hfss.set_active_design(name)
Change the active design to another design.
Parameters:
    name
        Name of the design to make active.

That's my latest frustration. :-)

With the help of fwlfHS I have made great progress.

JIm

jpotter0, 2022-11-25:
I finally succeeded at figuring out how to set the active design. I made a separate loop that just created the designs. Then I am able to select them by name inside the second loop. Amazing!

jpotter0, 2022-11-25:
It turns out that

design_name = "ihdtl"+str(cellnum)
aedtapp.set_active_design(design_name)

is all it needs in the loop that iterates cellnum. If the design isn't there it gets created.

The latest mystery is that I did something that makes the assign_symmetry not work. It is ignored. No error messages. I need to backtrack and see what changed,

Jim

closed 2022-11-26 by jpotter0