ansys/pyaedt #2027: Problem: How to change the properties of Object3D?

Jasonflash1996, 2022-11-23:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

After building a cylinder with the following code, is there any way to change its properties, such as radius?
```
from pyaedt import Maxwell3d
M3D = Maxwell3d()
M3D.modeler.create_cylinder(cs_axis="Z",
    position= [0,0,0],
    radius= 10,
    height= 10,
    name= "MagnetCylinder",
    matname= "TDK_PC95_25cel"
)
```
For example, change the radius of a cylinder named "MagnetCylinder" from 10 to 20. I have checked the API reference of pyaedt.modeler.Model3D.Modeler3D, and there is no applicable method.

### Steps To Reproduce

1. Create a Object3D like box, cylinder…
2. Change the attributes of Object3D, such as the length, width and height of the box, the radius of the cylinder, and so on.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

Before submitting the issue
 I have searched among the existing issues
 I am using a Python virtual environment
Description of the bug
I have a client A and a server B, both of which are Windows10 systems. AnsysEM and Pyaedt are installed on the server B. Currently, I'm using client-side VScode to connect to the server through SSH.
Operation on the server B:

from pyaedt.common_rpc import pyaedt_service_manager
pyaedt_service_manager()
After running the above code, the console displays:

pyaedt INFO: Starting the server on port 17878 on DESKTOP-CJLMS6V.
INFO:Global:Starting the server on port 17878 on DESKTOP-CJLMS6V.
INFO:SERVICEMANAGER/17878:server started on [222.20.85.183]:17878
Operation on the client A:
I created a new python script on the server through client-side VScode and write in the script:

from pyaedt import Maxwell3d
from pyaedt.common_rpc import create_session
cl1 = create_session("DESKTOP-CJLMS6V")
cl1.aedt(port=17878, non_graphical=False)
model= Maxwell3d(machine="DESKTOP-CJLMS6V", port=17878)
However, after running the above code, an AEDT dialog box pops up on the server side, displaying: "Unexpected command line argument '17878' ignored", with a "OK" button. After clicking OK, the graphic interface of AEDT is opened and a blank new project Project1 is displayed. There is no maxwell3d design in this project.
What puzzles me is the client-side VScode console shows:

pyaedt INFO: Created new session on port 19937
INFO:Global:Created new session on port 19937
Service has started on port 17878
pyaedt INFO: using existing logger.
INFO:Global:using existing logger.
pyaedt INFO: Launching PyAEDT outside AEDT with CPython and PythonNET.
INFO:Global:Launching PyAEDT outside AEDT with CPython and PythonNET. 
pyaedt INFO: AEDT installation Path D:\AnsysEM\v221\Win64.
INFO:Global:AEDT installation Path D:\AnsysEM\v221\Win64.
[... 26 lines cut ...]

Steps To Reproduce
Operation on the server B:

from pyaedt.common_rpc import pyaedt_service_manager
pyaedt_service_manager()
Operation on the client A:

Open VScode on the client side and use ssh to connect to the server side remotely.
Create a new python script file on the server side through VScode.
Write the following code in the file:
from pyaedt import Maxwell3d
from pyaedt.common_rpc import create_session
cl1 = create_session("DESKTOP-CJLMS6V")
cl1.aedt(port=17878, non_graphical=False)
model= Maxwell3d(machine="DESKTOP-CJLMS6V", port=17878)
Run the script.
Which Operating System are you using?
Windows

Which Python version are you using?
3.9

Installed packages
absl-py==0.15.0 aiohttp aiosignal alabaster anaconda-client anaconda-project anyio appdirs==1.4.4 argon2-cffi argon2-cffi-bindings arrow arviz astroid astropy asttokens astunparse==1.6.3 async-timeout atomicwrites==1.4.0 attrs Automat autopep8 Babel backcall backports.functools-lru-cache backports.tempfile backports.weakref==1.0.post1 bcrypt beautifulsoup4 binaryornot bitarray bkcharts==0.2 black==19.10b0 bleach bokeh boto3 botocore Bottleneck brewer2mpl==1.4.1 brotlipy==0.7.0 cachetools certifi==2022.9.24 cffi cftime chardet charset-normalizer clang==5.0 click cloudpickle clr-loader==0.2.4 clyent==1.2.2 colorama colorcet comtypes==1.1.10 conda-content-trust conda-pack conda-package-handling conda-repo-cli conda-verify==3.4.2 constantly==15.1.0 convertdate==2.4.0 cookiecutter cryptography cssselect==1.1.0 cupy-cuda112==10.5.0 cycler Cython cytoolz==0.11.0 daal4py==2021.5.0 dask datashader datashape==0.5.4 debugpy decorator defusedxml diff-match-patch distributed docutils entrypoints ephem==4.1.3 et-xmlfile==1.1.0 executing fastjsonschema fastrlock==0.8 filelock flake8 Flask flatbuffers==1.12 fonttools==4.25.0 frozenlist fsspec future gast==0.4.0 gensim glob2 google-api-core google-auth google-auth-oauthlib==0.4.6 google-cloud-core google-cloud-storage google-crc32c google-pasta==0.2.0 google-resumable-media googleapis-common-protos greenlet grpcio h5py HeapDict hijri-converter==2.2.4 holoviews hvplot hyperlink idna imagecodecs imageio imagesize importlib-metadata incremental inflection==0.5.1 iniconfig intake intervaltree ipykernel ipython ipython-genutils ipywidgets isort itemadapter itemloaders itsdangerous jdcal jedi Jinja2 jinja2-time jmespath joblib json5 jsonschema jupyter jupyter-client jupyter-console jupyter-core jupyter-server jupyterlab jupyterlab-pygments jupyterlab-server jupyterlab-widgets keras==2.6.0 Keras-Preprocessing==1.1.2 keyring kiwisolver korean-lunar-calendar==0.3.1 lazy-object-proxy libarchive-c llvmlite==0.38.0 locket lxml Markdown MarkupSafe matplotlib matplotlib-inline mccabe==0.6.1 menuinst mistune mkl-fft==1.3.1 mkl-random mkl-service==2.4.0 mock mpmath==1.2.1 msgpack multidict multipledispatch munkres==1.1.4 mypy-extensions==0.4.3 navigator-updater==0.2.1 nbclassic nbclient nbconvert nbformat nest-asyncio netCDF4 networkx nltk nose notebook numba numexpr numpy numpydoc oauthlib==3.2.0 olefile openpyxl opt-einsum==3.3.0 packaging pandas pandocfilters panel param paramiko parsel parso partd pathlib pathspec==0.7.0 patsy==0.5.2 pep8==1.7.1 pexpect pickleshare Pillow==9.0.1 pkginfo plotly pluggy plumbum==1.8.0 pooch==1.6.0 poyo prettyplotlib==0.1.7 prometheus-client prompt-toolkit Protego protobuf==3.19.1 psutil ptyprocess pure-eval py pyaedt==0.6.27 pyasn1 pyasn1-modules==0.2.8 pycodestyle pycosat==0.6.3 pycparser pyct pycurl==7.44.1 PyDispatcher==2.0.5 pydocstyle pyerfa pyflakes Pygments PyHamcrest PyJWT pylint pyls-spyder==0.4.0 PyMeeus==0.5.11 PyNaCl pyodbc pyOpenSSL pyparsing pyreadline==2.1 pyrsistent PySocks pystan==2.19.1.1 pytest==7.1.1 python-dateutil python-lsp-black python-lsp-jsonrpc==1.0.0 python-lsp-server==1.2.4 python-slugify python-snappy pythonnet==3.0.1 pytz==2021.3 pyvista==0.37.0 pyviz-comms PyWavelets pywin32==305 pywin32-ctypes pywinpty PyYAML==6.0 pyzmq QDarkStyle qstylizer QtAwesome qtconsole QtPy queuelib==1.5.0 regex requests requests-file requests-oauthlib==1.3.1 rope rpyc==5.0.1 rsa Rtree ruamel-yaml-conda s3transfer scikit-image scikit-learn scikit-learn-intelex==2021.20220215.102710 scipy scooby==0.7.0 Scrapy seaborn Send2Trash service-identity setuptools-git==1.2 sip==4.19.13 six smart-open sniffio snowballstemmer sortedcollections sortedcontainers soupsieve sparselandtools==1.0.1 Sphinx sphinxcontrib-applehelp sphinxcontrib-devhelp sphinxcontrib-htmlhelp sphinxcontrib-jsmath sphinxcontrib-qthelp sphinxcontrib-serializinghtml spyder spyder-kernels SQLAlchemy stack-data statsmodels==0.13.2 sympy tables==3.6.1 tabulate==0.8.9 tb-nightly==2.11.0a20221006 TBB==0.2 tblib tenacity tensorflow-estimator==2.9.0 tensorflow-gpu==2.6.0 termcolor==1.1.0 terminado testpath text-unidecode textdistance threadpoolctl three-merge tifffile tinycss tldextract toml tomli toolz tornado tqdm traitlets Twisted twisted-iocpsupport typed-ast typing_extensions ujson Unidecode urllib3 vtk==9.2.2 w3lib watchdog wcwidth webencodings==0.5.1 websocket-client Werkzeug widgetsnbextension win-inet-pton win-unicode-console==0.5 wincertstore==0.2 wrapt wslink==1.9.1 xarray xarray-einstats xlrd XlsxWriter xlwings==0.24.9 yapf yarl zict==2.0.0 zipp zope.interface

Samuelopez-ansys, 2022-11-23:
@Jasonflash1996  Which AEDT version are you using? From 2022R2, you can use the GRPC approach which is native to AEDT. 

If you want to use the 'legacy' approach, I think you should just use the following command:

Samuelopez-ansys, 2022-11-24:
@Jasonflash1996 Which AEDT version are you using?

If you can please try 2022R2, Because you could use GRPC natively in AEDT. Follow the steps in the Documentation:
https://aedt.docs.pyansys.com/Getting_started/ClientServer.html

About changing the Radius, you could use the Oriented AEDT Programming:

cl1 = create_session("MADSLOPEZ1")
cl1.aedt(port=17878, non_graphical=False)

model = Maxwell3d(machine="MADSLOPEZ1", port=17878)

cyl = model.modeler.create_cylinder(cs_axis="Z",
    position= [0,0,0],
    radius= 10,
    height= 10,
    name= "MagnetCylinder",
    matname= "TDK_PC95_25cel"
)

model.odesign.GetChildObject("3D Modeler").GetChildObject(cyl.name).GetChildObject("CreateCylinder:1").Set_Radius("20mm")

maxcapodi78, 2022-12-02:
@Jasonflash1996 Client Serve works on starting from AEDT 22R2. I see this in your log:
aedt INFO: AEDT installation Path D:\AnsysEM\v221\Win64.

Jasonflash1996, 2022-12-02:
@maxcapodi78 @Samuelopez-ansys Thank you for your comment! After I upgraded to 22R2, the problem was solved.

closed 2022-12-02 by Jasonflash1996