ansys/pyaedt #2030: Bug located in ...

felipeescastro, 2022-11-23:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hi, when trying to export an aedb to ipc2581, I get the following:

> In [1]: from pyaedt import Edb
> In [2]: inputfile = r"D:\aaSupport\test_six.aedb\edb.def"
> In [3]: edb = Edb(edbpath=inputfile,edbversion="2022.2")
> In [4]: edb.export_to_ipc2581(r"D:\aaSupport\test_six_ipc.xml","millimeter")
> pyaedt info: Export IPC 2581 is starting. This operation can take a while.
> pyaedt info: Error exporting IPC 2581.
> pyaedt info: Sequence contains no elements
>    at System.Linq.Enumerable.First[TSource](IEnumerable`1 source)
>    at EdbLib.IPC8521.IPCExporter.AddLayerSpecs(CadHeader CadHeader, Layout layout, String units)
>    at EdbLib.IPC8521.IPCExporter.BuildIPC2581(Layout l...
> Out[5]: False
> 2022-11-23 13:17:57,941 - INFO - Starting IPC2581 Export
> 2022-11-23 13:17:57,941 - INFO - Exporting Content Section
> 2022-11-23 13:17:57,941 - INFO - Exporting Layer Section

Test file can be shared internally to Ansys. Please reach out to get the file.
Kind regards,
Felipe

### Steps To Reproduce

Run command lines In [1] to In [4] in the bug description using database provided by me. 

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

aiohttp==3.8.3
aiosignal==1.2.0
alabaster==0.7.12
anyio==3.6.1
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
arrow==1.2.3
astroid==2.12.11
async-timeout==4.0.2
asynctest==0.13.0
atomicwrites==1.4.1
attrs==22.1.0
autopep8==1.6.0
Babel==2.10.3
backcall==0.2.0
bcrypt==4.0.1
beautifulsoup4==4.11.1
binaryornot==0.4.4
black==22.10.0
bleach==5.0.1
certifi==2022.9.24
cffi==1.15.1
chardet==5.0.0
charset-normalizer==2.1.1
click==8.1.3
cloudpickle==2.2.0
clr-loader==0.2.3
colorama==0.4.5
cookiecutter==2.1.1
cryptography==38.0.1
cycler==0.11.0
debugpy==1.6.3
decorator==5.1.1
defusedxml==0.7.1
diff-match-patch==20200713
dill==0.3.5.1
docutils==0.17.1
entrypoints==0.4
fastjsonschema==2.16.2
flake8==4.0.1
fonttools==4.37.4
frozenlist==1.3.1
idna==3.4
imageio==2.22.1
imagesize==1.4.1
importlib-metadata==4.2.0
importlib-resources==5.10.0
inflection==0.5.1
intervaltree==3.1.0
ipycanvas==0.13.1
ipyevents==2.0.1
ipykernel==6.16.0
ipython==7.34.0
ipython-genutils==0.2.0
ipyvtklink==0.2.3
ipywidgets==7.7.2
isort==5.10.1
jaraco.classes==3.2.3
jedi==0.18.1
jellyfish==0.9.0
Jinja2==3.1.2
jinja2-time==0.2.0
json5==0.9.10
jsonschema==4.16.0
jupyter-core==4.11.1
jupyter-server==1.21.0
jupyter_client==7.4.2
jupyterlab==3.4.8
jupyterlab-pygments==0.2.2
jupyterlab-widgets==1.1.1
jupyterlab_server==2.16.0
keyring==23.9.3
kiwisolver==1.4.4
lazy-object-proxy==1.7.1
MarkupSafe==2.1.1
matplotlib==3.5.3
matplotlib-inline==0.1.6
mccabe==0.6.1
mistune==2.0.4
more-itertools==8.14.0
multidict==6.0.2
mypy-extensions==0.4.3
nbclassic==0.4.5
nbclient==0.7.0
nbconvert==7.2.1
nbformat==5.7.0
nest-asyncio==1.5.6
notebook==6.5.1
notebook-shim==0.1.0
numpy==1.21.6
numpydoc==1.5.0
packaging==21.3
pandas==1.3.5
pandocfilters==1.5.0
paramiko==2.11.0
parso==0.8.3
pathspec==0.10.1
pexpect==4.8.0
pickleshare==0.7.5
Pillow==9.2.0
pkgutil_resolve_name==1.3.10
platformdirs==2.5.2
pluggy==1.0.0
plumbum==1.8.0
prometheus-client==0.15.0
prompt-toolkit==3.0.31
psutil==5.9.2
ptyprocess==0.7.0
pyaedt==0.6.20
pycodestyle==2.8.0
pycparser==2.21
pydocstyle==6.1.1
pyflakes==2.4.0
Pygments==2.13.0
pylint==2.15.4
pyls-spyder==0.4.0
PyNaCl==1.5.0
pyparsing==3.0.9
PyQt5==5.15.7
PyQt5-Qt5==5.15.2
PyQt5-sip==12.11.0
PyQtWebEngine==5.15.6
PyQtWebEngine-Qt5==5.15.2
pyrsistent==0.18.1
python-dateutil==2.8.2
python-lsp-black==1.2.1
python-lsp-jsonrpc==1.0.0
python-lsp-server==1.5.0
python-slugify==6.1.2
pythonnet==3.0.0.post1
pytoolconfig==1.2.2
pytz==2022.4
pyvista==0.36.1
pywin32==304
pywin32-ctypes==0.2.0
pywinpty==2.0.8
PyYAML==6.0
pyzmq==24.0.1
QDarkStyle==3.0.3
qstylizer==0.2.2
QtAwesome==1.1.1
qtconsole==5.3.2
QtPy==2.2.1
requests==2.28.1
rope==1.3.0
rpyc==5.0.1
Rtree==1.0.1
scooby==0.6.0
Send2Trash==1.8.0
six==1.16.0
sniffio==1.3.0
snowballstemmer==2.2.0
sortedcontainers==2.4.0
soupsieve==2.3.2.post1
Sphinx==4.3.2
sphinxcontrib-applehelp==1.0.2
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==2.0.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.5
spyder==5.3.3
spyder-kernels==2.3.3
terminado==0.16.0
text-unidecode==1.3
textdistance==4.5.0
three-merge==0.1.1
tinycss2==1.1.1
toml==0.10.2
tomli==2.0.1
tomlkit==0.11.5
tornado==6.2
traitlets==5.4.0
typed-ast==1.5.4
typing_extensions==4.4.0
ujson==5.5.0
urllib3==1.26.12
vtk==9.2.2
watchdog==2.1.9
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.4.1
whatthepatch==1.0.2
widgetsnbextension==3.6.1
wrapt==1.14.1
wslink==1.8.4
yapf==0.32.0
yarl==1.8.1
zipp==3.9.0

Samuelopez-ansys, 2022-11-23:
@svandenb-dev  @ring630

ring630, 2022-11-23:
@felipeescastro, I think the problem is Pyaedt.Edb accepts .aedb not .def.
 inputfile = r"D:\aaSupport\test_six.aedb\edb.def" ->  inputfile = r"D:\aaSupport\test_six.aedb"

felipeescastro, 2022-11-24:
@ring630 well spotted! It worked with the path up to .aedb. This issue can be closed. Then as ER, it would be good to accept the path up to edb.def as well to be aligned with electronics API.

closed 2022-11-24 by felipeescastro

ring630, 2022-11-28:
@felipeescastro supported :D