ansys/pyaedt #2039: Example 03_5G_antenna_example does not work.

MaxJPRey, 2022-11-25:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I tried to launch the FullDocumentation workflow and we have a bug within the example `03_5G_antenna_example.py`.
At some point it is calling `create_coax_port()` in padstack.py and to get back the `port_name`.

Then an error occur:
`TypeError: No method matches given arguments for GetLayerRange: ()`

The issue is that `create_coax_port` is expecting an "EDB" object and not a "PyAEDT" object anymore.
This is due to the fact that `place_padstack` is returning a "PyAEDT" object and not an "EDB" one.

### Steps To Reproduce

N/A

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

N/A

maxcapodi78, 2022-12-02:
Fixed and released in 0.6.32

closed 2022-12-02 by maxcapodi78

MaxJPRey, 2022-12-03:
Thanks @maxcapodi78.