ansys/pyaedt #2043: Why is a working code segment no longer working?

jpotter0, 2022-11-26:
The code below worked a few days ago. Now it does not work. I am successfully identifying the faces I wish ro use to set boundary conditions, but now the assign_symmetry function is ignored.

    obj_name = Cell[cellnum]

    faces = obj_name.faces_on_bounding_box

    sym_name1 = "sym1-cell" + str(cellnum)
    sym_name2 = "sym2-cell" + str(cellnum)
    sym_name3 = "sym3-cell" + str(cellnum)
    face_id1 = faces[7]
    face_id2 = faces[8]
    face_id3 = faces[2]
    aedtapp.assign_symmetry(face_id1, symmetry_name=sym_name1, is_perfect_e=False)
    aedtapp.assign_symmetry(face_id2, symmetry_name=sym_name2, is_perfect_e=False)
    aedtapp.assign_symmetry(face_id3, symmetry_name=sym_name3, is_perfect_e=False)

I traced into "aedtapp.assign_symmetry" and found "_functionhandler_wrapper (user function)." Instead of returning "result", it returned "false".

The question now is why "if not settings.enable_error_handler" is not returning true.

jpotter0, 2022-11-27:
I'm now trying to generate a simple program to focus on the assign_symmetry issue.

Here's what I've written so far.

"""Test"""
# %% Imports

import os
import pyaedt
from pyaedt import desktop
from pyaedt import Hfss

new_project = create_new_project =r'C:\Jims Files\PYScripts\test'
pyaedt.Hfss(projectname=new_project, designname="test1)",solution_type="Eigenmode")
version = "2022.2"
aedtapp: Hfss=Hfss()
non_graphical = os.getenv("PYAEDT_NON_GRAPHICAL","False").lower() in ("true","1","t")

# %% Code

X1 = 0.0
Y1 = 0.0
X2 = 100.0
y2 =  50.0
cell_points = [[X1,Y1,0.0],[X1,Y2,0.0],[X2,Y2,0.0],[X2,Y1,0.0],[X1,Y1,0.0]]
cell_name = cell
Cell = aedtapp.modeler.create_polyline(cell_points,segment_type=["Line","Line","Line","Line"],cover_surface=True,name=cell_name)
Cell_outer = aedtapp.modeler.sweep_along_vector(cell_name,[0.0,0.0,75.0])
obj_name = Cell
faces = obj_name.faces_on_bounding_box

aedtapp.release_desktop()
print("End of script")

Initially, I was debugging typos. When I got that cleaned up I run into a major snag.

I get an error message saying there is no module named 'pyaedt'.

C:\Users\jpott\PycharmProjects\pythonProject1\venv\Scripts\python.exe C:\Users\jpott\PycharmProjects\pythonProject1\main.py 
Traceback (most recent call last):
  File "C:\Users\jpott\PycharmProjects\pythonProject1\main.py", line 5, in <module>
    import pyaedt
ModuleNotFoundError: No module named 'pyaedt'

Process finished with exit code 1

This has been working in my larger code. What is going on?

Samuelopez-ansys, 2022-11-28:
Hi @jpotter0,

I think the issue is you are passing an ID, but the method expects a list of IDs, please see the documentation:

https://aedt.docs.pyansys.com/API/_autosummary/pyaedt.Hfss.assign_symmetry.html#assign-symmetry

About bypassing pyaedt, yes you can combine both, for example, if you have:

app = Hfss()

oDesign = app.odesign

And then use the same recorded commands.

Finally, regarding the installation, for now, PyAEDT is not integrated into the AEDT installation, so the only way is using pip install pyaedt, so you need to install it on your Python interpreter.

Let me know if it is clear.
Samuel

jpotter0, 2022-11-28:
> <https://github.com/pyansys/pyaedt/issues/2043#issuecomment-1328725889>view
> it on GitHub, or

Samuel,

Thanks for the response.

I have tried putting more than one ID in the list 
with the same result. I have also found that 
enclosing a single item in brackets works if a 
list is required. The code as is generates no 
error messages. It is simply ignored. It returns 
False instead of a result. The real mystery is 
that I swear it worked at one time exactly as 
written. something has turned off its ability to respond to assign_symmetry.

I will try your suggestion to access the AEDT script.

Best regards,

Jim

Samuelopez-ansys, 2022-11-28:
@jpotter0 Please could you try to create a simple code for trying to reproduce the error (open HFSS, create a box, and create symmetry)? If it fails, please share it here and I will debug it.

jpotter0, 2022-11-28:
Samuel,

I have exactly such a code. I will send it in a 
minute. I am trying out the access to the AEDT methods,

Jim

James M. Potter, PhD, President
JP Accelerator Works, Inc.
2245 47th Street
Los Alamos, NM 87544

TEL: 505-690-8701

jpotter0, 2022-11-28:
Samuel,

A test code is attached. I had been messing with it and had to restore it.

Jim

"""Script for IHDTL HFSS design using PyAEDT."""

# %% Imports

import math
import os
import pyaedt

from pyaedt import Desktop
from pyaedt import Hfss

# %% Code

new_project = create_new_project = r'C:\Jims Files\PY Scripts\TEST.aedt'

pyaedt.Hfss(projectname=new_project, designname="test1", solution_type="Eigenmode")
ersion = "2022.2"

aedtapp: Hfss = Hfss()
###############################################################################
# Set non-graphical mode

non_graphical = os.getenv("PYAEDT_NON_GRAPHICAL", "False").lower() in ("true", "1", "t")

###############################################################################
# Launch AEDT
# ~~~~~~~~~~~
# Launch AEDT 2022 R2 in graphical mode.

d = Desktop("2022.2", non_graphical=False, new_desktop_session=True)

###############################################################################
# Launch HFSS

# %% Codesettings

x1 = 0.0
y1 = 0.0
x2 = 100.0
y2 = 75.0
z1 = 50.0

cell_points = [[x1, y1, 0.0], [x2, y1, 0.0], [x2, y2, 0.0],[x1, y2, 0.0], [x1, y1, 0.0]]
cell_name = "cell"
Cell = aedtapp.modeler.create_polyline(cell_points, segment_type=["Line", "Line", "Line", "Line"],
                                            cover_surface=True, name=cell_name)

Cell1 = aedtapp.modeler.sweep_along_vector(cell_name, [0.0, 0.0, z1])

obj_name = Cell

faces = obj_name.faces_on_bounding_box

sym_name1 = "sym1-cell"
sym_name2 = "sym2-cell"
sym_name3 = "sym3-cell"
face_id1 = faces[0]
face_id2 = faces[2]
face_id3 = faces[3]
aedtapp.assign_symmetry(face_id1, symmetry_name=sym_name1, is_perfect_e=False)
aedtapp.assign_symmetry(face_id2, symmetry_name=sym_name2, is_perfect_e=False)
aedtapp.assign_symmetry(face_id3, symmetry_name=sym_name3, is_perfect_e=False)

aedtapp.release_desktop()

print("End of script")

Samuelopez-ansys, 2022-11-28:
HI @jpotter0,

I have detected a bug for Eigenmode solution type, I will fix the bug. Once it is fixed, you just need to:

sym_name1 = "sym1-cell"
sym_name2 = "sym2-cell"
sym_name3 = "sym3-cell"
face_id1 = faces[0]
face_id2 = faces[2]
face_id3 = faces[3]
aedtapp.assign_symmetry([face_id1, face_id2, face_id3], symmetry_name=sym_name1, is_perfect_e=False)

jpotter0, 2022-11-28:
Samuel,

Maybe that's why it worked before. At first, while creating the geometry, the solution method defaulted to modal. When I got closer to wanting to run the solver, I paid more attention to setting up the filename and the design names.

I'm looking forward to getting the fix.

Best regards,

Jim

closed 2022-11-28 by maxcapodi78

Samuelopez-ansys, 2022-11-28:
@jpotter0 

Using the batch file is just run:

pyaedt_with_IDE.bat -u

About the repo, please contact Alex Moody, and he could help you because it is not pyaedt related.

jpotter0, 2022-11-29:
Samuel,

My update is to v0.6.28

How to I get v0.6.30?

Jim

James M. Potter, PhD, President
JP Accelerator Works, Inc.
2245 47th Street
Los Alamos, NM 87544

TEL: 505-690-8701

PR #2047 (2022-11-28): Add Eigenmode option
Close #2043