ansys/pyaedt #2056: Bug in showing nets in HFSS3DLayout

SHF101202021, 2022-11-29:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

## Problem
**The following codes will not show the whole nets**

These are my nets
[image]

Here is the code
```python

hfss = Hfss3dLayout(specified_version='2022.2', 
                    projectname='certain project',
                    designname='certain design')

# This print will not show all the nets
print(hfss.modeler.nets)

```

Here are how console shows
[image]

## Reason
> PyAEDT get net names by parsing the group **<Power/Ground>** and **Non Power/Ground**

[image]

## Suggestion
When getting all nets of the design, use ```GetNets()``` instead
[image]

I suggest the code can be modified as the following:

```python
    @property
    def nets(self):
        """Nets.

        Returns
        -------
        dict[str, :class:`pyaedt.modeler.object3dlayout.Nets3DLayout`]
            Nets Dictionary.

        """

        if self._nets:
            return self._nets

        objs = self.modeler.oeditor.GetNets()
        for obj in objs:
            self._nets[obj] = Nets3DLayout(self, obj)
        return self._nets

```

### Steps To Reproduce

As mentioned in the above.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

Not Important here

SHF101202021, 2022-11-29:
I can try to push the code myself (If you had some instruction, that will be better.  If not, I can figure out myself)
But I'm not sure if I had the authority to do that

SHF101202021, 2022-11-29:
@gmalinve 

Please check **Net Block** in AEDT
The reason you can get the ```len(nets) = len(signal_nets) + len(power_nets)``` is that you have both ```<Power/Ground>``` and ```Non Power/Ground``` Net Group

For my project, ```Non Power/Ground``` net group is missing so that my nets cannot show signal_nets
[image]

gmalinve, 2022-11-30:
Hi @SHF101202021 ,
A PR to fix this issue has just been created.
Thanks for reporting!

closed 2022-11-30 by gmalinve