ansys/pyaedt #2059: Bug located in create_voltage_source_on_pin_group / create_current_source_on_pin_group

pyAnsysUser, 2022-11-29:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When defining a negative source (positive ground) in addition to a positive source, the negative source does not appear in the source list of the created project

### Steps To Reproduce

```
import os
from pyaedt import Edb,examples
edb_path = os.path.join(long_edb_path, 'Temp\edb\Galileo.aedb')
edb_portTest = Edb(edbpath=edb_path, edbversion="2022.2")
edb_portTest.save_edb_as(os.path.join(long_edb_path, 'Temp\edb_test\Galileo.aedb'))

# Get ground pins
GND_U2A5_pin_num = []
GND_U2A5_pins = edb_portTest.core_components.get_pin_from_component('U2A5', netName='GND')
[GND_U2A5_pin_num.append(nam.GetName()) for nam in GND_U2A5_pins]
GND_U2A5_group = 'GND_U2A5'

GND_J2L1_pin_num = []
GND_J2L1_pins = edb_portTest.core_components.get_pin_from_component('J2L1', netName='GND')
[GND_J2L1_pin_num.append(nam.GetName()) for nam in GND_J2L1_pins]
GND_J2L1_group = 'GND_J2L1'

# Get supply pins 
V3p3_U2A5_pin_num = []
V3p3_U2A5_pins = edb_portTest.core_components.get_pin_from_component('U2A5', netName='V3P3_S3')
[V3p3_U2A5_pin_num.append(nam.GetName()) for nam in V3p3_U2A5_pins]
V3p3_U2A5_group = 'V3P3_U2A5'

V3p3_J2L1_pin_num = []
V3p3_J2L1_pins = edb_portTest.core_components.get_pin_from_component('J2L1', netName='V3P3_S3')
[V3p3_J2L1_pin_num.append(nam.GetName()) for nam in V3p3_J2L1_pins]
V3p3_J2L1_group = 'V3P3_J2L1'

# Get supply pins - Assume 1.5V negative potential
V1p5_U2A5_pin_num = []
V1p5_U2A5_pins = edb_portTest.core_components.get_pin_from_component('U2A5', netName='V1P5_S0')
[V1p5_U2A5_pin_num.append(nam.GetName()) for nam in V1p5_U2A5_pins]
V1p5_U2A5_group = 'V1P5_U2A5'

V1p5_J2L1_pin_num = []
V1p5_J2L1_pins = edb_portTest.core_components.get_pin_from_component('J2L1', netName='V1P5_S0')
[V1p5_J2L1_pin_num.append(nam.GetName()) for nam in V1p5_J2L1_pins]
V1p5_J2L1_group = 'V1P5_J2L1'

# Create Pin Groups
V3p3_U2A5_PG = edb_portTest.core_siwave.create_pin_group('U2A5',V3p3_U2A5_pin_num,V3p3_U2A5_group)
V1p5_U2A5_PG = edb_portTest.core_siwave.create_pin_group('U2A5',V1p5_U2A5_pin_num,V1p5_U2A5_group)
GND_U2A5_PG = edb_portTest.core_siwave.create_pin_group('U2A5',GND_U2A5_pin_num,GND_U2A5_group)

V3p3_J2L1_PG = edb_portTest.core_siwave.create_pin_group('J2L1',V3p3_J2L1_pin_num,V3p3_J2L1_group)
V1p5_J2L1_PG = edb_portTest.core_siwave.create_pin_group('J2L1',V1p5_J2L1_pin_num,V1p5_J2L1_group)
GND_J2L1_PG = edb_portTest.core_siwave.create_pin_group('J2L1',GND_J2L1_pin_num,GND_J2L1_group)

# Create Voltage Source
edb_portTest.core_siwave.create_voltage_source_on_pin_group(V3p3_J2L1_group, GND_J2L1_group, 3.3, 0, "Vpos_3p3")
# Create Voltage Source - Assume 1.5V negative potential
edb_portTest.core_siwave.create_voltage_source_on_pin_group(GND_J2L1_group,V1p5_J2L1_group, 1.5, 0, "Vneg_1p5") #this does not work
# edb_portTest.core_siwave.create_voltage_source_on_pin_group(V1p5_J2L1_group, GND_J2L1_group, -1.5, 0, "Vneg_1p5") #this works

#Create Current Source
edb_portTest.core_siwave.create_current_source_on_pin_group(V3p3_U2A5_group, GND_U2A5_group, 1.0, 0, "Ipos_3p3")
#Create Current Source - Assume negative current
edb_portTest.core_siwave.create_current_source_on_pin_group(GND_U2A5_group,V1p5_U2A5_group, 1.0, 0, "Ineg_1p5") #this does not work
# edb_portTest.core_siwave.create_current_source_on_pin_group(V1p5_U2A5_group,GND_U2A5_group, -1.0, 0, "Ineg_1p5") #this works

settings = edb_portTest.core_siwave.get_siwave_dc_setup_template()
settings.accuracy_level = 0
settings.use_dc_custom_settings = True
settings.name = "myDCIR_4"
settings.neg_term_to_ground = "Vpos_3p3"
# settings.pos_term_to_ground = "Vneg_1p5" # I don't think this is needed
edb_portTest.core_siwave.add_siwave_dc_analysis(settings)

edb_portTest.save_edb()
edb_portTest.solve_siwave()

edb_portTest.close_edb()

```

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

package list not public

ring630, 2022-12-02:
Hi, @pyAnsysUser. Which line fails to create the source in your example?

ring630, 2022-12-02:
@pyAnsysUser fixed.

closed 2022-12-02 by maxcapodi78

pyAnsysUser, 2022-12-02:
when running the lines above, I get a new error:
 AttributeError: 'EdbSiwave' object has no attribute 'get_siwave_dc_setup_template'

Did something change around this attribute?

pyAnsysUser, 2022-12-05:
After reviewing the changelog, I was able to modify and run the code snippet, however the desired outcome does not appear. I will review the changes in further detail to see if this functionality is met.

ring630, 2022-12-05:
@pyAnsysUser , there is a bug. We are working on it.