ansys/pyaedt #2071: Use pyansys/actions in the workflow.

MaxJPRey, 2022-12-01:
### Description of the feature

💡 **Idea**
With the recent release of [pyansys/actions], we could upgrade the CI/CD workflows.

In addition these actions introduced the multi-version documentation feature, meaning that we can get rid of our stable and developers doc repositories.

[pyansys/actions]: https://github.com/pyansys/actions

PipKat, 2022-12-02:
@MaxJPRey We'd still have the "release" doc and the "in-main/production" doc though, yes?

MaxJPRey, 2022-12-02:
Yes @PipKat , but all gathered under the same website.

MaxJPRey, 2023-02-17:
Both the `FullDocumentation` and `Nightly Documentation` are now taking advantage of the `pyansys/actions`.

closed 2023-02-17 by MaxJPRey