ansys/pyaedt #2079: When I try to analyze HFSS Eigenmode geometry on pyaedt I get an error message

jpotter0, 2022-12-02:
The error message is "Invalid Setup 'ihdrl1', but if I run that same setup manually it works fine. What is happening?

Samuelopez-ansys, 2022-12-02:
PyAEDT can not detect if the Analysis runs correctly or not because the analyze_setup method can not retrieve this info in the AEDT API.

Please could you share an snippet code to reproduce the error?

@jpotter0

jpotter0, 2022-12-02:
> <https://github.com/pyansys/pyaedt/issues/2079#issuecomment-1335502573>view
> it on GitHub, or

Thanks for the reply, Samuel.
Here''s the code:

num = 0
for num in range(ncells):
cellnum = num + 1
design_name = "ihdtl"+str(cellnum)
aedtapp.set_active_design(design_name)
r_cell = 0.001 * rcell[cellnum]
aedtapp["r_cell"] = str(r_cell)
variations = aedtapp.available_variations.nominal_w_values_dict
variations["r_cell"] = ["1"]
running = aedtapp.analyze_setup("ihdtl"+str(cellnum))
post = aedtapp.post
data = post.get_solution_data("re(Mode(1))")
var = data.data_real()
fcell[cellnum] = 1E-6 * var[1]
print (f'{str(var),+" "+str(cellnum)+" 
"+str(rcell[cellnum])+" "+str(fcell[cellnum])}')

Here's the setup code, which seems to work. As I 
said the setup works when I run it manually.

oModule = oDesign.GetModule("AnalysisSetup")
oModule.InsertSetup("HfssEigen",
[
"NAME:Setup"+str(cellnum),
"MinimumFrequency:=" , "300MHz",
"NumModes:=" , 1,
"MaxDeltaFreq:=" , 0.03,
"ConvergeOnRealFreq:=" , True,
"MaximumPasses:=" , 10,
"MinimumPasses:=" , 1,
"MinimumConvergedPasses:=" , 1,
"PercentRefinement:=" , 30,
"IsEnabled:=" , True,
[
"NAME:MeshLink",
"ImportMesh:=" , False
],
"BasisOrder:=" , 1,
"DoLambdaRefine:=" , True,
"DoMaterialLambda:=" , True,
"SetLambdaTarget:=" , False,
"Target:=" , 0.2,
"UseMaxTetIncrease:=" , False

Thanks,

Jim

jpotter0, 2022-12-02:
Samuel,

I figured it out. The function didn't wan the 
name of the design. It wanted the name of the setup, "Setup1".

Thanks again!

Jim

jpotter0, 2022-12-02:
I get a error message about the index being out of range on

fcell[cellnum] = 1E-6 * var[0]

var is a list and the frequency is the 0-th 
element in the list so I son't understand why the error message.

Jim

jpotter0, 2022-12-04:
Figured out how to make it work.

closed 2022-12-04 by jpotter0