ansys/pyaedt #2085: I need to get the integral of the z component of the ComplexMag_E along a lne

jpotter0, 2022-12-05:
I can do this manually in the Fields Calculator. This is the effective voltage across an accelerating cell gap. I need to divide by the path length to get the average E field. Then I can scale the solution energy to normalize my data to 1 MV/m.

When I do it manually the result ends up at the top of the stack. When I try to use the script in my pyaedt program I can get everything I copied to an HFSS script to run, but I don't see how to get the result as a numerical answer. Is there an easier way in pyaedt?

I tried using oModule.CalcOp("Value")
but that resulted in an error message:

  File "C:\Jims Files\PY Scripts\ihdtl4.py", line 448, in <module>
    vv = oModule.CalcOp("Value")
  File "<COMObject GetModule>", line 2, in CalcOp
pywintypes.com_error: (-2147352567, 'Exception occurred.', (0, None, None, None, 0, -2147024365), None)

Samuelopez-ansys, 2022-12-05:
Hi @jpotter0 ,

You will need to create a NamedExpression and then get the solution data, you can not get the data directly from the Fields Calculator, for example:

Fields = M3D.odesign.GetModule("FieldsReporter")
Fields.EnterQty("B")
Fields.CalcOp("ScalarZ")
Fields.EnterScalar(1000)
Fields.CalcOp("*")
Fields.CalcOp("Smooth")
Fields.AddNamedExpression("Bz", "Fields")

Then:

solutions = M3D.post.get_solution_data(
    expressions="mag(Bz)",
    report_category="Fields",
    context="Line_AB",
    variations=variations,
    primary_sweep_variable="Distance",
)

In 'solutions' you will have the result from the Fields Calculator. Please if it is not clear, share with us a code to reproduce your workflow (with the Fields Calculator operations) and I will try to get the data you need.

jpotter0, 2022-12-05:
I found that I could request that package M3D be installed. It appeared to install and a line import M3D appeared near the other import lines.

However, when I run the program I get the following error message:

Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'M3D'

Somehow it must not have the right path for M3D.

How do I fix that?

jpotter0, 2022-12-05:
Your instructions are pretty clear. It looks to me like I will get a list of 1000 values of Ez (what I need) along my line axis1, but I have to get past the M3D hurdle first.

jpotter0, 2022-12-05:
I found the installation of M3D:

[image]

jpotter0, 2022-12-06:
Major confusion over the definition of M3D. It is supposed to be just Hfss() but I mistook it for a module. There is a module named M3D and I was able to install it.

closed 2022-12-06 by jpotter0