ansys/pyaedt #2086: HFSS 3D Layout: Support 'tech file' when GDS file is imported not only 'xml file'

ngkazu, 2022-12-05:
### Description of the feature

Some of users use 'tech file' when GDS file is imported.
Now, PyAEDT support only 'xml file' as layer data but I would like you to be able to set 'tech file' as control_file. 

### Steps for implementing the feature

- Hfss3dLayout.import_gds(
     gds_path, 
     aedb_path=None, 
     control_file=None, 
     set_as_active=True, 
     close_active_project=False)

if  extension of control_file is 'xml':
   read xml file...
elif extension of control_file is 'tech':
   read xml file...

or  add 'contorol_file_type' as argument of method.
if control_file_type is 'xml':
  read xml file...
elif contorol_file_type is 'tech:
  read tech file...

maxcapodi78, 2022-12-12:
@ngkazu Is the tech file looking always like the following? 
Because I noticed that there is no way from API to impor it so I should translate to xml and then use it. but before implementing I would like to be sure about uniqness of the tech format

UNITS um

/---------------------------------------------------------
/ L#	Label	Color	Elevation	Thickness
/---------------------------------------------------------
14	Bump 	purple		0		100

closed 2022-12-13 by maxcapodi78

ngkazu, 2022-12-14:
@maxcapodi78 Thank you for your replay.
Yes, you are correct.
That's the format same as described in HFSS3D Layout Help, right?
It's ok to use that format.

PR #2131 (2022-12-13): Support for TECH files in GDS and DXF imports in HFSS 3D Layout
Fix #2086.