ansys/pyaedt #2113: Bug located in Maxwell2d.variable_manager

ansRMorin, 2022-12-08:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

The evaluated value retrieved from the variable_manager[variable].evaluated_value is not correct. For variable = 'electric_period' shown in the image below, I'm expecting the evaluated value to be '0.0029411765' as displayed in EDT. 
[image]
However, when I use the script below, the evaluated value returned eval_value = "0.00294117647058824**rad**" 
[image]

### Steps To Reproduce

1. Create a .aedt project with 3 properties:

[image]
[image]
[image]

2. Run the following code

[image]

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

aiohttp==3.8.3
aiosignal==1.3.1
appdirs==1.4.4
async-timeout==4.0.2
asynctest==0.13.0
attrs==22.1.0
certifi==2022.9.24
cffi==1.15.1
charset-normalizer==2.1.1
clr-loader==0.2.4
cycler==0.11.0
fonttools==4.38.0
frozenlist==1.3.3
idna==3.4
imageio==2.22.4
kiwisolver==1.4.4
matplotlib==3.5.3
multidict==6.0.2
numpy==1.21.6
packaging==21.3
pandas==1.3.5
Pillow==9.3.0
plumbum==1.8.0
pooch==1.6.0
psutil==5.9.4
pyaedt==0.6.28
pycparser==2.21
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.6
pyvista==0.37.0
pywin32==305
requests==2.28.1
rpyc==5.0.1
scooby==0.7.0
six==1.16.0
typing-extensions==4.4.0
urllib3==1.26.12
vtk==9.2.2
wslink==1.9.1
yarl==1.8.1

gmalinve, 2022-12-09:
Hi @ansRMorin ,

Have you tried to use numeric_value or value instead of evaluated_value?
These two should give you back the expected number as a float.

[image]
[image]

Then you can use the round Math function to round the number.

ansRMorin, 2022-12-09:
Hi @gmalinve,
Thank you for the response. In fact, the issue shows up when I retrieve the units, which I'm expecting a unit of time (if solved explicitly) or an empty string to be returned (if read from the units column in Maxwell). The way the variable is defined we divide an angle (rad or deg) by rpm which is a rotation (=360deg or 2pi rad) by a time so the outcome should be a time.

maxcapodi78, 2022-12-12:
#2126

closed 2022-12-12 by maxcapodi78