ansys/pyaedt #2151: Add ...

dcrawforAtAnsys, 2022-12-15:
### Description of the feature

Enable extraction of field quantities during Maxwell (2D and 3D) transient simulation using the control program.  Here is the syntax in the native API to set up a python script for the control program
[image]

### Steps for implementing the feature

1. Ensure that control program is part of the solution setup
2. Pass information to the method: 
 - field quantity
 - output file name
 - analysis setup.  
 - It is desirable (not sure it is possible) to return the data in the method rather than writing to a text file.  
3. Run transient simulation. Field data will be written to a file.

### Useful links and references

Some relevant methods:

- https://aedt.docs.pyansys.com/API/_autosummary/pyaedt.maxwell.Maxwell2d.setup_ctrlprog.html
- https://aedt.docs.pyansys.com/API/_autosummary/pyaedt.maxwell.Maxwell2d.src_dir.html

Samuelopez-ansys, 2023-05-11:
@gmalinve Is this related to the new feature you added?

gmalinve, 2023-05-11:
@Samuelopez-ansys @dcrawforAtAnsys yep! I implemented it in the previous release :)
well.. actually it was an old method and I improved it.

Samuelopez-ansys, 2023-05-11:
Then this issue could be closed ? @gmalinve

gmalinve, 2023-05-11:
@Samuelopez-ansys @dcrawforAtAnsys I'll close this one and as agreed with Devin I'll open an issue to implement a complete example for control program in pyaedt.

closed 2023-05-11 by gmalinve