ansys/pyaedt #2211: set_solution_broadband cannot create valide setup

ring630, 2022-12-27:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

set_solution_broadband cannot create valide setup. 

### Steps To Reproduce

`import os
from pyaedt import Edb

appedb = Edb(edbversion="2022.2")
setup = appedb.create_hfss_setup("setup1")
setup.set_solution_broadband("1GHz", "2GHz")

for i in appedb.setups["setup1"].adaptive_settings.adaptive_frequency_data_list:
    print(i.adaptive_frequency)

os.chdir(r"C:\ansysdev\_aedt_workspace\draft")
appedb.save_edb_as("broadband.aedb")`

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

aiohttp==3.8.3
aiosignal==1.3.1
anyio==3.6.2
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
asttokens==2.2.0
async-timeout==4.0.2
attrs==22.1.0
Babel==2.11.0
backcall==0.2.0
beautifulsoup4==4.11.1
black==22.10.0
bleach==5.0.1
certifi @ file:///C:/b/abs_ac29jvt43w/croot/certifi_1665076682579/work/certifi
cffi==1.15.1
charset-normalizer==2.1.1
click==8.1.3
clr-loader==0.2.4
colorama==0.4.6
contourpy==1.0.6
cycler==0.11.0
debugpy==1.6.4
decorator==5.1.1
defusedxml==0.7.1
entrypoints==0.4
exceptiongroup==1.0.4
executing==1.2.0
fastjsonschema==2.16.2
fonttools==4.38.0
frozenlist==1.3.3
idna==3.4
imageio==2.22.4
iniconfig==1.1.1
ipykernel==6.17.1
ipython==8.7.0
ipython-genutils==0.2.0
jedi==0.18.2
Jinja2==3.1.2
json5==0.9.10
jsonschema==4.17.3
jupyter-server==1.23.3
jupyter_client==7.4.7
jupyter_core==5.1.0
jupyterlab==3.5.0
jupyterlab-pygments==0.2.2
jupyterlab_server==2.16.3
kiwisolver==1.4.4
MarkupSafe==2.1.1
matplotlib==3.6.2
matplotlib-inline==0.1.6
mistune==2.0.4
multidict==6.0.2
mypy-extensions==0.4.3
nbclassic==0.4.8
nbclient==0.7.2
nbconvert==7.2.5
nbformat==5.7.0
nest-asyncio==1.5.6
notebook==6.5.2
notebook_shim==0.2.2
numpy==1.23.5
packaging==21.3
pandas==1.5.2
pandocfilters==1.5.0
parso==0.8.3
pathspec==0.10.2
pickleshare==0.7.5
Pillow==9.3.0
platformdirs==2.5.4
pluggy==1.0.0
plumbum==1.8.0
pooch==1.6.0
prometheus-client==0.15.0
prompt-toolkit==3.0.33
psutil==5.9.4
pure-eval==0.2.2
pycparser==2.21
Pygments==2.13.0
pyparsing==3.0.9
pyrsistent==0.19.2
pytest==7.2.0
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.6
pyvista==0.37.0
pywin32==305
pywinpty==2.0.9
pyzmq==24.0.1
requests==2.28.1
rpyc==5.0.1
scooby==0.7.0
Send2Trash==1.8.0
six==1.16.0
sniffio==1.3.0
soupsieve==2.3.2.post1
stack-data==0.6.2
terminado==0.17.0
tinycss2==1.2.1
tomli==2.0.1
tornado==6.2
traitlets==5.6.0
urllib3==1.26.13
vtk==9.2.2
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.4.2
wincertstore==0.2
wslink==1.9.1
yarl==1.8.1

ring630, 2022-12-27:
@maxcapodi78 @svandenb-dev , I am not able to understand the behavior of edblib. many redundant frequency points are added automatically.

svandenb-dev, 2023-01-09:
@ring630 I took ownership if this issue. fyi edblib dll has been removed.

svandenb-dev, 2023-01-09:
@ring630 do you really need to use Aedt API ? because I think there is an option in EDB