ansys/pyaedt #2250: Nightly doc is not working anymore.

MaxJPRey, 2023-01-09:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

See the [nightly-docs](https://github.com/pyansys/pyaedt/actions/workflows/nightly-docs.yml) history. 

The documentation from the main branch is built properly but is not deployed.

[image]

### Steps To Reproduce

N/A.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

N/A.

jorgepiloto, 2023-01-09:
Related with https://github.com/pyansys/actions/pull/131

jorgepiloto, 2023-01-10:
Using [pyansys/actions@v3](actions.docs.pyansys.com/release/3.0/index.html) should fix this issue.

MaxJPRey, 2023-01-10:
@jorgepiloto exactly. I have launched the build with one of my branch embedding the v3.
Thanks @jorgepiloto for the new release of pyansys/actions.

closed 2023-02-17 by MaxJPRey