ansys/pyaedt #2257: Cannot start Desktop

Jan-HendrikPlank, 2023-01-10:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

When I try to start the Desktop using pyaedt on my Windows machine I get the following error message:

pyaedt INFO: using existing logger.
pyaedt INFO: Launching PyAEDT outside AEDT with CPython and PythonNET.
pyaedt INFO: AEDT installation Path C:\Program Files\AnsysEM\AnsysEM21.2\Win64.
pyaedt INFO: Launching AEDT with module PythonNET.

Traceback (most recent call last):
  File "py_aedt_based.py", line 3, in <module>
    with pyaedt.Desktop(
  File "<path_to_venv>\lib\site-packages\pyaedt\generic\design_types.py", line 479, in Desktop
    return app(
  File "<path_to_venv>\lib\site-packages\pyaedt\desktop.py", line 406, in __init__
    self._init_cpython(
  File "<path_to_venv>\lib\site-packages\pyaedt\desktop.py", line 672, in _init_cpython
    self.logger.info("{} Started with process ID {}.".format(version, proc[0]))
IndexError: list index out of range

I can see that the Electronics Desktop has been started. 

Debugging into the function where it crashes shows me that the list proc, as well as the lists processID and processID2 are empty. So for me it looks like for some reason it fails to fetch the process id.

### Steps To Reproduce

Using the starting example from the docs and execute if from within Pycharm in a fresh venv:

import pyaedt

with pyaedt.Desktop(
    specified_version="2021.2",
    non_graphical=False,
    new_desktop_session=True,
    close_on_exit=True,
    student_version=False,
):
    circuit = pyaedt.Circuit()

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

appdirs==1.4.4
certifi==2022.12.7
cffi==1.15.1
charset-normalizer==2.1.1
clr-loader==0.2.5
contourpy==1.0.6
cycler==0.11.0
fonttools==4.38.0
idna==3.4
imageio==2.24.0
kiwisolver==1.4.4
matplotlib==3.6.2
numpy==1.24.1
packaging==23.0
pandas==1.5.2
Pillow==9.4.0
plumbum==1.8.1
pooch==1.6.0
psutil==5.9.4
pyaedt==0.6.42
pycparser==2.21
pyparsing==3.0.9
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.7
pyvista==0.37.0
pywin32==305
requests==2.28.1
rpyc==5.3.0
scooby==0.7.0
six==1.16.0
urllib3==1.26.13
vtk==9.2.5

teng23, 2023-02-02:
Hi, I'm new to AEDT and I had the same problem. I changed the pyaedt to an older version and it worked. It seams that all older version before 0.6.44 are working.

aelsay96, 2023-02-06:
i faced the same problem and i tried to install older version as @teng23 said " the problem related to version  start from 0.6.44 "

maxcapodi78, 2023-04-17:
@pyaedt is supported from "2022.1". Anyway I've fixed a bug recently which allows to use it also in 2021.2. This doesn't guarantee that all methods in PyAEDT works in such release of AEDT but at least it works. I suggest to install latest version of pyaedt and retry

[image]