ansys/pyaedt #2281: pyaedt_with_IDE.bat Windows install produces error: ResolutionTooDeep: 2000000

em-jim, 2023-01-12:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Experiencing very long install times with errors when running the pyaedt_with_IDE.bat file.

The installer kept trying different versions of packages.

My Windows 11 laptop took 5 hours.  But the end I see an error: 
raise ResolutionTooDeep(max_rounds)
pip._vendor.resolvelib.resolvers.ResolutionTooDeep: 2000000
  
Here is the complete CMD terminal session outputs:
[long install.txt](https://github.com/pyansys/pyaedt/files/10405640/long.install.txt)

The installation did not complete.

### Steps To Reproduce

pyaedt 0.6.42

1. Download: pyaedt_with_IDE.bat 
from: https://aedt.docs.pyansys.com/release/0.6/Getting_started/Installation.html

2. (Here is a copy of the .bat file; remove .txt from .bat file below to use it or download it from the website listed above) 
[pyaedt_with_IDE.bat.txt](https://github.com/pyansys/pyaedt/files/10405708/pyaedt_with_IDE.bat.txt)

3. Open admin CMD terminal and run the  pyaedt_with_IDE.bat   

4. Selected "0=InstallOnly" when running the .bat file.  

The installation took 5 hours to finish; but produced an Error: ResolutionTooDeep: 2000000

CMD terminal outputs:
[long install.txt](https://github.com/pyansys/pyaedt/files/10405640/long.install.txt)

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

[pip freeze output.txt](https://github.com/pyansys/pyaedt/files/10405685/pip.freeze.output.txt)

i.e.
C:\work\pyAEDT>C:\Users\jrsherma\AppData\Roaming\pyaedt_env_ide\Scripts\python -m pip freeze
anyio==3.6.2
appdirs==1.4.4
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
attrs==22.2.0
Babel==2.11.0
backcall==0.2.0
beautifulsoup4==4.11.1
bleach==5.0.1
certifi==2022.12.7
cffi==1.15.1
charset-normalizer==2.1.1
clr-loader==0.2.5
colorama==0.4.6
cycler==0.11.0
debugpy==1.6.5
decorator==5.1.1
defusedxml==0.7.1
entrypoints==0.4
fastjsonschema==2.16.2
fonttools==4.38.0
idna==3.4
imageio==2.24.0
importlib-metadata==6.0.0
importlib-resources==5.10.2
ipycanvas==0.13.1
ipyevents==2.0.1
ipykernel==6.16.2
ipython==7.34.0
ipython-genutils==0.2.0
ipyvtklink==0.2.3
ipywidgets==7.7.2
jedi==0.18.2
Jinja2==3.1.2
json5==0.9.11
jsonschema==4.17.3
jupyter-server==1.23.4
jupyter_client==7.4.8
jupyter_core==4.12.0
jupyterlab==3.5.2
jupyterlab-pygments==0.2.2
jupyterlab-widgets==1.1.1
jupyterlab_server==2.18.0
kiwisolver==1.4.4
MarkupSafe==2.1.1
matplotlib==3.5.3
matplotlib-inline==0.1.6
mistune==2.0.4
nbclassic==0.4.8
nbclient==0.7.2
nbconvert==7.2.7
nbformat==5.7.3
nest-asyncio==1.5.6
notebook==6.5.2
notebook_shim==0.2.2
numpy==1.21.6
packaging==23.0
pandas==1.3.5
pandocfilters==1.5.0
parso==0.8.3
pickleshare==0.7.5
Pillow==9.4.0
pkgutil_resolve_name==1.3.10
plumbum==1.8.1
pooch==1.6.0
prometheus-client==0.15.0
prompt-toolkit==3.0.36
psutil==5.9.4
pyaedt==0.6.42
pycparser==2.21
Pygments==2.14.0
pyparsing==3.0.9
pyrsistent==0.19.3
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.7
pyvista==0.37.0
pywin32==305
pywinpty==2.0.10
pyzmq==25.0.0
requests==2.28.1
rpyc==5.3.0
scooby==0.7.0
Send2Trash==1.8.0
six==1.16.0
sniffio==1.3.0
soupsieve==2.3.2.post1
terminado==0.17.1
tinycss2==1.2.1
tomli==2.0.1
tornado==6.2
traitlets==5.8.1
typing_extensions==4.4.0
urllib3==1.26.14
vtk==9.2.5
wcwidth==0.2.5
webencodings==0.5.1
websocket-client==1.4.2
widgetsnbextension==3.6.1
zipp==3.11.0

em-jim, 2023-01-13:
I installed PyCharm 20.1.1 and python 3.7.9 and this works fine and installed fast.  

I added a couple libraries:
cpython
ipython
pyaedt
matplotlib
numpy
pyvista

pyaedt runs the examples files and opens AEDT.

em-jim, 2023-01-31:
The new .bat file for 0.6.45 is excellent!

This allow the user to integrate pyAEDT inside Electronics Desktop in the Project Tool menu.

This new implementation is much better than before.

Thank You

closed 2023-01-31 by em-jim

MaxJPRey, 2023-02-01:
Thanks @em-jim for the confirmation that it fixed the issue.