ansys/pyaedt #2354: Improve Setup class (or derived classes)

dcrawforAtAnsys, 2023-01-25:
### Description of the current limitations

Add an analyze method to the Setup class. This will simplify syntax and eliminate some redundancy, like passing the setup name to the hfss.analyze_nominal() method.

### Code sample expected

Usage:

hfss = pyaedt.Hfss(specified_version="2022.2")
setup = hfss.create_setup(setupname="Setup1", setuptype="HFSSDriven",
                          Frequency="10GHz", MaximumPasses=15)

setup.create_linear_count_sweep(
    unit="GHz",
    freqstart=9.5,
    freqstop=10.5,
    sweepname="sweep1",
    sweep_type="Interpolating",
    interpolation_tol=0.5,
    interpolation_max_solutions=255,
    save_fields=False,
)

setup.analyze()

### Useful links and references

https://aedt.docs.pyansys.com/release/0.6/API/_autosummary/pyaedt.hfss.Hfss.analyze_setup.html

dcrawforAtAnsys, 2023-01-26:
This is very easy to implement. The high level application is available to a Setup object as either ``.p_app`` or ``._app``. Then the ``app.analyze_setup()`` can be called from within the Setup class and the name of the setup passed as the first argument. All named arguments for ''analyze_setup()'' are passed through the method.

Samuelopez-ansys, 2023-03-23:
I will close this issue, the feature was implemented:
https://aedt.docs.pyansys.com/release/0.6/API/_autosummary/pyaedt.modules.SolveSetup.SetupHFSS.analyze.html

closed 2023-03-23 by Samuelopez-ansys