ansys/pyaedt #2358: Add method to read and change impedance multiplier to boundary module

fwlfHS, 2023-01-25:
### Description of the feature

Method to modify the impedance multiplier in case of symmetry boundary is missing in the boundary module.
(Or I haven't found it.)

It would be nice if there is a method to get the current multiplier and another one to set/change the value.

### Steps for implementing the feature

pyaedt.modules.Boundary.get_impedance_multiplier()
pyaedt.modules.Boundary.set_impedance_multiplier(value)

### Useful links and references

mod = hfss.odesign.GetModule('BoundarySetup')
mod.ChangeImpedanceMult(value)

Samuelopez-ansys, 2023-07-06:
Hi,

I will implement the set multiplier, the get multiplier is not accessible from the API, so for now it is a limitation.

closed 2023-07-07 by Samuelopez-ansys

fwlfHS, 2023-07-26:
@Samuelopez-ansys thank you for the implementation. However the naming convention is a bit misleading. The float value given is the multiplier not the impedance.
[image]
Can you please rename that variable. Thanks in advance.

Samuelopez-ansys, 2023-07-26:
Hi @fwlfHS ,

PyAEDT is open source, everyone can contribute, maybe this could be a good opportunity for you to do a first contribution to PyAEDT, if you don’t have time, I will try to refactor it in the next 3 weeks.

Please let me know if you are going to contribute.

PR #3236 (2023-07-07): Add impedance multiplier
Close #2358