ansys/pyaedt #2367: Add get face handler by id to 3D modeler

fwlfHS, 2023-01-26:
### Description of the feature

If I know the ID of an objects face, I would like to get the face handler by the ID. Similar to 
```python
from pyaedt import Hfss
hfss = Hfss()
region = hfss.modeler.get_object_from_name('Region')
face = region.bottom_face_x
face.center
# to be implemented:
face = hfss.modeler.get_face_by_id(38)
face.center
```

dcrawforAtAnsys, 2023-01-26:
By "handler" do you mean Class?  A face is not always planar, so the ``center`` property would not be so easy to define for non-planar surfaces and would even involve some overhead that could bog things down for complex models for planar faces.  If you share more information about why you need the face center, there may be another way to achieve your objective.

fwlfHS, 2023-01-27:
In my application I have the region with a symmetry boundary on a certain face, I'd like to get the symmetry "direction" by getting the normal vector of this face. **The face is planar in this case** and I can use the "handler" bottom_face_x if I know the direction of the symmetry condition. But If I don't know that, I can get the face ID by `hfss.boundaries[sym_id]['Faces']`, now I need to get the normal vector of the face with this ID, but I have no clue how to access the properties of this face, with only the ID. 
Well, I could loop over all region faces and compare their ID with the bottom/top x/y/z stuff, but this seems like a workaround to me.

dcrawforAtAnsys, 2023-01-27:
If you know the face_ID and the 3D object ID (or name of the 3D object) you could retrieve face objects with the ``.faces`` property of Modeler3D.  What you really need is to be able to retrieve the face object given the face ID.  This appears not to be exposed in the API.  That would solve your problem:
```python
normal = hfss.modeler.faces[face_ID].normal  # Not yet implemented. Would return an instance of 
```
I will look into this further, but in the meantime if you can retrieve the face object (instance of ``FacePrimitive`` the ``normal`` property should give you what you want.

dcrawforAtAnsys, 2023-01-27:
Here is a workound.  I'll implement a more elegant solution soon.  

Given the face_id, and the Object3D object, you can itereate through the ``faces``  and find the face whose ID matches your Face_ID.  Then use the ``normal`` property of the face object:
``` python
my_object = hfss.modeler.get_object_from_name('my_object')
for face in my_object.faces:
    if face.id == face_id:
        this_face = face
```
now ``this_face.normal`` should give you the face normal.

fwlfHS, 2023-01-27:
Thank you, that works for me currently!

bakroistvan, 2023-05-08:
@dcrawforAtAnsys is there any update on this?

gmalinve, 2023-08-24:
Hi @fwlfHS, @bakroistvan ,

Your request has been recently added and will be available in next pyaedt release.
You'll be able to access get_face_by_id from modeler.

Hope this helps,
Kind regards,

Giulia

closed 2023-08-24 by gmalinve