ansys/pyaedt #2382: Simplify syntax to access solution data for native post-processing in Python

dcrawforAtAnsys, 2023-01-29:
### Description of the feature

Syntax for expressions and solution in ``pyaedt.modules.PostProcessor.Reports`` is directly passed through to the native API without syntax checking and is non-intuitive and not "Pythonic".  Access to solution data should be simplified to provide easier and more intuitive access to solution data.

### Steps for implementing the feature

Solution data should be made accessible from the Setup class, pyaedt.modules.Setup or possibly the CommonSetup class since the solution data is always directly linked to the solution setup. Implementation needs further investigation.  Access to solution data will still go through the reporter in the native API.

### Useful links and references

- ``pyaedt.modules.PostProcessor.py``
- ``Reports class``

Samuelopez-ansys, 2023-10-24:
Hi @dcrawforAtAnsys ,

Any update regarding this issue?

Samuelopez-ansys, 2024-10-01:
With the new Visualization refactoring, this issue is solved.

closed 2024-10-01 by Samuelopez-ansys