ansys/pyaedt #2389: Add control_file to import_brd function as done below. Currently it does not have ability to select onl

vishak-venkatraman, 2023-01-30:
### Description of the feature

add a contol_file to import_brd:  this would enable selective net import rather than entire board.

### Steps for implementing the feature

```
  def import_brd(
        self, input_file, aedb_path=None, control_file=None, set_as_active=True, close_active_project=False
    ):  # pragma: no cover
        """Import a board file into HFSS 3D Layout and assign the stackup from an XML file if present.

        Parameters
        ----------
        input_file : str
            Full path to the board file.
        aedb_path : str, optional
            Full path to the AEDB file.
        set_as_active : bool, optional
            Whether to set the board file as active. The default is ``True``.
        close_active_project : bool, optional
            Whether to close the active project after loading the board file.
            The default is ''False``.
        Returns
        -------
        bool
            ``True`` when successful, ``False`` when failed.

        References
        ----------

        >>> oModule.ImportExtracta
        """
        return self._import_cad(input_file, "brd", aedb_path, control_file, set_as_active, close_active_project)

```

maxcapodi78, 2023-01-31:
@vishak-venkatraman I'll check if it is supported by anstranslator and if so I'll implement in next released

svandenb-dev, 2023-02-20:
@maxcapodi78 this is done ? can we close the issue ?

closed 2023-03-20 by vishak-venkatraman