ansys/pyaedt #2416: Bug located in ...no boundaries when opening existing Maxwell model

HDLPY, 2023-02-02:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hello Team,

      When existing Maxwell model with winding and motion, I find they are not in boundaries yet. Please see below.
      We must run pyaedt commands to assign them, and then I could find them in the boundaries list.
 
      When I open existing model with winding and motion, I could not find any boundaries automatically in 2023R1.
 

[image]
[Py1.zip](https://github.com/pyansys/pyaedt/files/10573773/Py1.zip)

### Steps To Reproduce

Please check the attach files. 
1). Just run py to open or load existing Maxwell model.
2). Evaluate M2D.boundaries, and couldn't find any boundary. 

Thanks.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.7

### Installed packages

2to3==1.0
appdirs==1.4.4
atomicwrites==1.4.1
attrs==22.1.0
backcall==0.2.0
certifi==2022.9.14
cffi==1.15.1
chardet==3.0.4
clr-loader==0.2.5
colorama==0.4.5
cycler==0.11.0
decorator==5.1.1
execnet==1.9.0
future==0.16.0
googleapis-common-protos==1.52.0
grpcio==1.26.0
grpcio-status==1.26.0
idna==2.7
imageio==2.25.0
importlib-metadata==4.12.0
iniconfig==1.1.1
ipython==7.34.0
jedi==0.18.2
jsonrpclib @ file:///C:/ANSYSDev/TFSAgent/_work/1/s/CPythonExtensionModules/jsonrpclib/0.1.3/jsonrpclib-0.1.3.tar.gz
jsonschema==3.2.0
keyring @ file:///C:/ANSYSDev/TFSAgent/_work/1/s/CPythonExtensionModules/keyring/23.5.0/keyring-23.5.0.tar.gz
kiwisolver==1.4.4
matplotlib==3.1.1
matplotlib-inline==0.1.6
more-itertools==8.14.0
nine==0.3.4
nose @ file:///C:/ANSYSDev/TFSAgent/_work/1/s/CPythonExtensionModules/nose/1.3.7/nose-1.3.7.tar.gz
numpy @ file:///C:/ANSYSDev/TFSAgent/_work/1/s/CPythonExtensionModules/numpy/1.21.1/numpy-1.21.1.zip
numpy-stl==1.7.1
packaging==21.3
pandas==1.3.5
parso==0.8.3
pickleshare==0.7.5
Pillow==9.2.0
pluggy==0.13.1
plumbum==1.8.1
polib==1.0.7
pooch==1.6.0
prompt-toolkit==3.0.36
protobuf==3.18.1
psutil==5.8.0
py==1.11.0
pyaedt==0.6.45
pybind11==2.10.0
pycparser==2.21
Pygments==2.14.0
pyparsing==3.0.9
pyreadline==2.1
pyrsistent==0.18.1
pytest==6.0.1
pytest-forked==1.4.0
pytest-xdist==1.31.0
python-certifi-win32==1.6
python-dateutil==2.8.2
python-utils==1.6.2
pythonnet==3.0.1
pytz==2019.3
pyvista==0.37.0
pywin32==301
pywin32-ctypes==0.2.0
PyYAML==5.3.1
requests==2.20.1
rpyc==5.3.0
scipy==1.7.0
SCons==4.0.1
scooby==0.7.1
setuptools-scm==7.0.5
simplejson @ file:///C:/ANSYSDev/TFSAgent/_work/1/s/CPythonExtensionModules/simplejson/3.16.0/simplejson-3.16.0.tar.gz
sip==6.5.1
six==1.16.0
toml==0.10.2
tomli==2.0.1
traitlets==5.8.1
twisted==15.3.0
typing-extensions==4.3.0
urllib3==1.24.3
virtualenv @ file:///C:/ANSYSDev/TFSAgent/_work/1/s/CPythonExtensionModules/virtualenv/12.1.1/virtualenv-12.1.1.tar.gz
vtk==9.2.5
wcwidth==0.2.6
wrapt==1.14.1
wxPython==4.1.1
zipp==3.8.1
zope.interface==4.1.2

gmalinve, 2023-02-15:
Hi @HDLPY ,

Thanks for you feedback.
There was actually an issue in the rename method, it will be fixed in a new PR and be available in the next release.

Giulia

closed 2023-02-17 by gmalinve

HDLPY, 2023-02-17:
Thanks a lot.