ansys/pyaedt #2420: Maxwell 2D Initial mesh Settings

tizianrot, 2023-02-03:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

While try to enhance initial mesh settings in a 2D electrostatic, cannot:
1 - change the slider value to a higher value through the command:
     M2D.mesh.assign_initial_mesh_from_slider(level=10)
     the script runs without performing the change
2- if I need to switch to manual settings, I can set manual setting, but I do not find the property to access surface deviation values. I tried to reproduce what recorded in Iron Python scripts but with no success:
    mesh_ims = M2D.mesh.initial_mesh_settings
    mesh_ims['CurvedSurfaceApproxChoice'] = 'ManualSettings' # until here ok
    #mesh_ims['SurfDevChoice'] = 2 # this appears in the IronPython, I tried with and without
    mesh_ims['SurfDev'] = '0.01mm' 
    M2D.mesh.initial_mesh_settings.update() # manual setting activated, but value stays unchecked and greyed out

Note: in 2023R1 classic mesher for 2D was disabled, so it may be that the the method is not updated to the new window where some options are no longer available

### Steps To Reproduce

see description above

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

--

gmalinve, 2023-02-21:
Hi @tizianrot ,

There are actually 9 available values for the initial mesh settings.
[image]
Available level values are from 1 to 9.

tizianrot, 2023-02-22:
Hi @gmalinve,

thank this is helpful, how about the manual settings when I want to define i.e. a specific  surface deviation value?

Thanks and warm regards

gmalinve, 2023-02-22:
Hi @tizianrot,

To assign the surface deviation you should use assign_surface_mesh_manual() method where you can specify as input the surf_dev input. You can specify float number with units here.

Please do let me know if that works for you.

Giulia

tizianrot, 2023-02-22:
Hallo @gmalinve , the slider function works fine.
 The other function works also fine with object assignment, actually I meant in the initial settings where it is applied to the whole model. 
Thank you

closed 2023-03-01 by gmalinve