ansys/pyaedt #2429: Add pydocstyle to the pre-commit

MaxJPRey, 2023-02-06:
### Description of the feature

In the pre-commit, we could use the `pydocstyle` hook.
This will create warnings where documentation is missing when adding new features.

### Steps for implementing the feature

Here is some code to write in the `.pre-commit-config.yaml`.

```
- repo: https://github.com/pycqa/pydocstyle
  rev: 6.1.1
  hooks:
  - id: pydocstyle
    additional_dependencies: [toml]
    exclude: "tests/"
```

Samuelopez-ansys, 2023-04-14:
@MaxJPRey Is this still planned to implement?

MaxJPRey, 2023-04-14:
@Samuelopez-ansys Yes, I will have to do it iteratively in #2836 .
I am expecting a lot of errors and warnings at first.

PR #2836 (2023-04-21): Add pydocstyle to pre-commit config.
Fix #2429
closed 2023-04-21 by MaxJPRey