ansys/pyaedt #2436: Add setter method to EDBPadstackInstance.pin_number

Marcel0000, 2023-02-07:
### Description of the feature

It would be great to have a setter method for pin numbers to EDBPadstackInstance.pin_number.

This should enable the possibility to rename pins easily without trigger pin removal from the component, as it is currently the case.

### Useful links and references

https://aedt.docs.pyansys.com/release/0.6/EDBAPI/_autosummary/pyaedt.edb_core.edb_data.padstacks_data.EDBPadstackInstance.pin_number.html#pyaedt.edb_core.edb_data.padstacks_data.EDBPadstackInstance.pin_number

ring630, 2023-11-13:
@Marcel0000 I have done many experiments. It is a technical limitation.

Marcel0000, 2023-11-15:
@ring630 alright, thanks for checking this out!

Samuelopez-ansys, 2023-11-15:
@ring630 If it is a technical limitation of EDB, could we close this issue?

ring630, 2023-11-15:
@Samuelopez-ansys Yes. I am closing it now.

closed 2024-01-08 by ring630