ansys/pyaedt #2442: create_port_on_pin_group mismatch while executing multiple times

raindigoa, 2023-02-09:
Hi,

here is a bug that happened to me many times,

when I tried to set ports by the command "create_port_on_pin_group"

there are always some ports mismatching and they have no rule(the mismatch grp pairs change every time)

for example, we have 

the pos_pin_groups = [f'pos_pin_grp{i}' for i in range(1,31)]
the neg_pin_groups = [f'neg_pin_grp{i}' for i in range(1,31)]
and the port_names = [f'port_{i}' for i in range(1,31)]

and we execute the command:

for i in range(0,len(port_names)):
edb.core_siwave.create_port_on_pin_group(pos_pin_groups[i],neg_pin_groups[i],name=port_names[i])

after that we save the edb and import it into the tool like siwave, 

you can always check the tab"ports" in the circuit element parameters and find something like

port1 50 pos_net neg_net pos_pin_grp_**2** neg_pin_grp_1
port2 50 pos_net neg_net pos_pin_grp_**2** neg_pin_grp_**2**
...
port29 50 pos_net neg_net pos_pin_grp_**28** neg_pin_grp_**29**
port30 50 pos_net neg_net pos_pin_grp_**30** neg_pin_grp_**29**

The same issue happens even when I write it line by line, or even add a sleep(5) between line and line. I think there must be something wrong.

Besides, I also want to highlight the was to display the ports paired by pos/neg pin groups in SIWAVE GUI are quite different when you create the ports by siwave manually and by pyaedt.

When you create the ports by siwave manually, you will see the circles represent the pos/neg pin group are aligned with the center.
However, when you create the ports by pyaedt (EDB) and import it into the siwave, the circles located around the corresponding pin groups "randomly", this would confuse the user and make them spend time on checking the port settings again and again.

That's all I would like to report in this post, please fix them, thank you.

svandenb-dev, 2023-02-13:
Hi @raindigoa,
thanks for submitting this feedback. I will have a look to fix it.

svandenb-dev, 2023-04-19:
@raindigoa one point about the port rendering in Siwave Vs Aedt. The EDB is escatly the same between tools. The difference comes the EDB import from Ansys products. However electrical connection is identical, this is purely graphical rendering.