ansys/pyaedt #2447: Bug located in ...

donlee115, 2023-02-10:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

Hi,

I am a quite beginner to Pyaedt, and I run some provided example files.
However, the code cannot find the aedt2022 R1 version. 
I used the Linux environment, and the Ansys HFSS 2022 R1 version is installed in /usr/pack/ansys-2022-wi/AnsysEM/v221.
Although I add the path, the code still cannot find the aedt.
In this case, how can I resolve this issue.
The screenshot is attached below.
[image]

### Steps To Reproduce

run code ....

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.9

### Installed packages

# Name                    Version                   Build  Channel
_anaconda_depends         2022.05                  py39_0  
_ipyw_jlab_nb_ext_conf    0.1.0            py39h06a4308_1  
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
aiohttp                   3.8.1            py39h7f8727e_1  
aiosignal                 1.2.0              pyhd3eb1b0_0  
alabaster                 0.7.12             pyhd3eb1b0_0  
anaconda                  custom                   py39_1  
anaconda-client           1.11.0           py39h06a4308_0  
anaconda-navigator        2.2.0            py39h06a4308_0  
anaconda-project          0.11.1           py39h06a4308_0  
anyio                     3.5.0            py39h06a4308_0  
appdirs                   1.4.4              pyhd3eb1b0_0  
argon2-cffi               21.3.0             pyhd3eb1b0_0  
argon2-cffi-bindings      21.2.0           py39h7f8727e_0  
arrow                     1.2.3            py39h06a4308_0  
ase                       3.22.1                   pypi_0    pypi
astroid                   2.11.7           py39h06a4308_0  
astropy                   5.1              py39h7deecbd_0  
asttokens                 2.0.5              pyhd3eb1b0_0  
async-timeout             4.0.2            py39h06a4308_0  
atomicwrites              1.4.0                      py_0  
attrs                     21.4.0             pyhd3eb1b0_0  
automat                   20.2.0                     py_0  
autopep8                  1.6.0              pyhd3eb1b0_1  
babel                     2.9.1              pyhd3eb1b0_0  
backcall                  0.2.0              pyhd3eb1b0_0  
backports                 1.1                pyhd3eb1b0_0  
backports.functools_lru_cache 1.6.4              pyhd3eb1b0_0  
backports.tempfile        1.0                pyhd3eb1b0_1  
backports.weakref         1.0.post1                  py_1  
bcrypt                    3.2.0            py39h5eee18b_1  
beautifulsoup4            4.11.1           py39h06a4308_0  
binaryornot               0.4.4              pyhd3eb1b0_1  
bitarray                  2.5.1            py39h5eee18b_0  
bkcharts                  0.2              py39h06a4308_1  
black                     22.6.0           py39h06a4308_0  
blas                      1.0                         mkl  
[... 437 lines cut ...]
webencodings              0.5.1            py39h06a4308_1  
websocket-client          0.58.0           py39h06a4308_4  
werkzeug                  2.0.3              pyhd3eb1b0_0  
wget                      1.21.3               h0b77cf5_0  
whatthepatch              1.0.2            py39h06a4308_0  
wheel                     0.37.1             pyhd3eb1b0_0  
widgetsnbextension        3.5.2            py39h06a4308_0  
wrapt                     1.14.1           py39h5eee18b_0  
wurlitzer                 3.0.2            py39h06a4308_0  
xarray                    0.20.1             pyhd3eb1b0_1  
xlrd                      2.0.1              pyhd3eb1b0_0  
xlsxwriter                3.0.3              pyhd3eb1b0_0  
xlwt                      1.3.0            py39h06a4308_0  
xz                        5.2.6                h5eee18b_0  
yaml                      0.2.5                h7b6447c_0  
yapf                      0.31.0             pyhd3eb1b0_0  
yarl                      1.8.1            py39h5eee18b_0  
zeromq                    4.3.4                h2531618_0  
zfp                       0.5.5                h295c915_6  
zict                      2.1.0            py39h06a4308_0  
zipp                      3.8.0            py39h06a4308_0  
zlib                      1.2.13               h5eee18b_0  
zope                      1.0              py39h06a4308_1  
zope.interface            5.4.0            py39h7f8727e_0  
zstd                      1.5.2                ha4553b6_0

gmalinve, 2023-02-14:
Hi @donlee115 ,

version 2022.1 is not supported on Linux.
In Linux AEDT is launch only in grpc mode which has been introduced from version 2022.2.

Thanks!

Giulia

gmalinve, 2023-02-15:
Hi @donlee115 ,

Please do let me know if my explanation was self explanatory enough so I can close this issue.

Thanks,

Giulia

donlee115, 2023-02-15:
Thank you, Giulia. I will try with 2022 R2 version later. 

Best,
Dongwon

closed 2023-02-15 by donlee115