ansys/pyaedt #2451: Bug located in ...Specified version 2022.2 is not known

kunal-3196, 2023-02-12:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I've downloaded Ansys 2022 R2,when running pyedt, throwing below error
    assert specified_version in self.version_keys, "Specified version {} is not known".format(specified_version)

AssertionError: Specified version 2022.2 is not known

### Steps To Reproduce

Run below code

import os
from matplotlib import pyplot as plt
import numpy as np
import pyaedt

###############################################################################
# Set non-graphical mode
# ~~~~~~~~~~~~~~~~~~~~~~
# Set non-graphical mode, ``"PYAEDT_NON_GRAPHICAL"`` is needed to generate
# documentation only.
# You can set ``non_graphical`` either to ``True`` or ``False``.

non_graphical = os.getenv("PYAEDT_NON_GRAPHICAL", "False").lower() in ("true", "1", "t")

###############################################################################
# Launch AEDT with Circuit
# ~~~~~~~~~~~~~~~~~~~~~~~~
# Launch AEDT 2022 R2 in graphical mode with Circuit.

cir = pyaedt.Circuit(projectname=pyaedt.generate_unique_project_name(),
                     specified_version="2022.2",
                     new_desktop_session=True,
                     non_graphical=non_graphical
                     )

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

raceback (most recent call last):

  File C:\PyAnsys\MAPDL\Circuit_Transient_issue.py:27 in <module>
    cir = pyaedt.Circuit(projectname=pyaedt.generate_unique_project_name(),

  File ~\anaconda3\lib\site-packages\pyaedt\circuit.py:125 in __init__
    FieldAnalysisCircuit.__init__(

  File ~\anaconda3\lib\site-packages\pyaedt\application\AnalysisNexxim.py:41 in __init__
    Analysis.__init__(

  File ~\anaconda3\lib\site-packages\pyaedt\application\Analysis.py:100 in __init__
    Design.__init__(

  File ~\anaconda3\lib\site-packages\pyaedt\application\Design.py:170 in __init__
    desktop = Desktop(

  File ~\anaconda3\lib\site-packages\pyaedt\desktop.py:400 in __init__
    self._main.student_version, version_key, version = self._set_version(specified_version, student_version)

  File ~\anaconda3\lib\site-packages\pyaedt\desktop.py:528 in _set_version
    assert specified_version in self.version_keys, "Specified version {} is not known".format(specified_version)

AssertionError: Specified version 2022.2 is not known

Samuelopez-ansys, 2023-03-17:
Hi @kunal-3196 ,

Do you have installed Ansys Electronics Desktop (AEDT) 2022R2? If you install the Ansys package, it is only installing Mechanical and Workbench, AEDT is not included.
If you have it, please try to create a clean virtual environment, do again pip install pyaedt, and try again. Let us know which pyaedt version is using:

import pyaedt
pyaedt.__version__

And which OS you are using.

Before running an example, just try:

import pyaedt

app = pyaedt.Hfss()
app.release_desktop()

gmalinve, 2023-03-27:
Hi @kunal-3196 ,

was this suggestion helpful?
In case it was, could you please let us know so we can close the issue?

Thanks!

Giulia

kunal-3196, 2023-04-01:
Thanks resolved