ansys/pyaedt #2460: Two tests in test_21_Circuit are failing

isaacansys, 2023-02-13:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

These two tests are failing on main as of d1830e889934269ff70b0c043b2853ada3b4b84c:

```
(venv) PS C:\git\Ansys\pyaedt> pytest _unittest/test_21_Circuit.py::TestClass::test_06b_add_3dlayout_component _unittest/test_21_Circuit.py::TestClass::test_07a_push_excitation
=================================================================================== test session starts ====================================================================================
platform win32 -- Python 3.9.12, pytest-7.2.1, pluggy-1.0.0
rootdir: C:\git\Ansys\pyaedt, configfile: pytest.ini
plugins: cov-4.0.0, xdist-3.2.0
collected 2 items

_unittest\test_21_Circuit.py FF                                                                                                                                                       [100%]

========================================================================================= FAILURES ========================================================================================= 
________________________________________________________________________ TestClass.test_06b_add_3dlayout_component _________________________________________________________________________ 

self = <_unittest.test_21_Circuit.TestClass object at 0x0000019DA0EE0C10>

    def test_06b_add_3dlayout_component(self):
        myedb = self.aedtapp.modeler.schematic.add_subcircuit_3dlayout("Galileo_G87173_204")
        assert type(myedb.id) is int
        ports = myedb.pins
        tx = ports
        rx = ports
        insertions = ["dB(S({},{}))".format(i.name, j.name) for i, j in zip(tx, rx)]
>       assert self.aedtapp.post.create_report(
            insertions,
            self.aedtapp.nominal_adaptive,
            plotname="Insertion Losses",
            plot_type="Rectangular Plot",
            report_category="Standard",
            subdesign_id=myedb.id,
        )
E       AssertionError: assert False
E        +  where False = <bound method PostProcessorCommon.create_report of <pyaedt.modules.PostProcessor.CircuitPostProcessor object at 0x0000019DC0F9BD90>>(['dB(S(J3B2.3.USBH2_DP_CH,J3B2.3.USBH2_DP_CH))', 'dB(S(J3B2.2.USBH2_DN_CH,J3B2.2.USBH2_DN_CH))', 'dB(S(J2B3-2,J2B3-2))'], '', plotname='Insertion Losses', plot_type='Rectangular Plot', report_category='Standard', subdesign_id=60)
E        +    where <bound method PostProcessorCommon.create_report of <pyaedt.modules.PostProcessor.CircuitPostProcessor object at 0x0000019DC0F9BD90>> = <pyaedt.modules.PostProcessor.CircuitPostProcessor object at 0x0000019DC0F9BD90>.create_report
E        +      where <pyaedt.modules.PostProcessor.CircuitPostProcessor object at 0x0000019DC0F9BD90> = <pyaedt.circuit.Circuit object at 0x0000019D9F253B20>.post
E        +        where <pyaedt.circuit.Circuit object at 0x0000019D9F253B20> = <_unittest.test_21_Circuit.TestClass object at 0x0000019DA0EE0C10>.aedtapp
E        +    and   '' = <pyaedt.circuit.Circuit object at 0x0000019D9F253B20>.nominal_adaptive
E        +      where <pyaedt.circuit.Circuit object at 0x0000019D9F253B20> = <_unittest.test_21_Circuit.TestClass object at 0x0000019DA0EE0C10>.aedtapp
E        +    and   60 = <pyaedt.modeler.circuits.object3dcircuit.CircuitComponent object at 0x0000019DC0D77D30>.id

_unittest\test_21_Circuit.py:107: AssertionError
---------------------------------------------------------------------------------- Captured stdout setup ----------------------------------------------------------------------------------- 
pyaedt INFO: using existing logger.
pyaedt INFO: Launching PyAEDT outside AEDT with CPython and PythonNET.
pyaedt INFO: AEDT installation Path C:\Program Files\AnsysEM\v221\Win64.
pyaedt INFO: Launching AEDT with module PythonNET.
pyaedt INFO: Ansoft.ElectronicsDesktop.2022.1 Started with process ID 18428.
pyaedt INFO: pyaedt v0.7.dev0
pyaedt INFO: Python version 3.9.12 (tags/v3.9.12:b28265d, Mar 23 2022, 23:52:46) [MSC v.1929 64 bit (AMD64)]
pyaedt INFO: AEDT file load (threaded) time: 0.17030119895935059
pyaedt INFO: Project Galileo_t21 has been opened.
pyaedt INFO: No consistent unique design is present. Inserting a new design.
pyaedt INFO: Added design 'Circuit Design_WLV' of type Circuit Design.
pyaedt INFO: Aedt Objects initialized
pyaedt INFO: Variable Manager initialized
pyaedt INFO: Design Loaded
pyaedt INFO: Successfully loaded project materials !
pyaedt INFO: Materials Loaded
pyaedt INFO: AEDT file load (threaded) time: 0.04727625846862793
pyaedt INFO: Project differential_pairs has been opened.
pyaedt INFO: Active Design set to Circuit1
pyaedt INFO: Aedt Objects initialized
[... 279 lines cut ...]
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.6
pyvista==0.38.1
pywin32==305
PyYAML==6.0
requests==2.28.1
rpyc==5.3.0
scikit-learn==1.2.1
scipy==1.10.0
scooby==0.7.0
shapely==2.0.1
six==1.16.0
SRTM.py==0.3.7
stack-data==0.6.2
threadpoolctl==3.1.0
tomli==2.0.1
traitlets==5.9.0
urllib3==1.26.12
utm==0.7.0
virtualenv==20.19.0
vtk==9.2.2
wcwidth==0.2.6
wslink==1.9.1
yarl==1.8.1

gmalinve, 2023-02-14:
Hi  @isaacansys ,

Thanks for you feedback.
We have just tested the latest main and we didn't get any failed test.
Is your main up to date?

Giulia

gmalinve, 2023-02-14:
https://github.com/pyansys/pyaedt/actions/runs/4165397291/jobs/7208338299

isaacansys, 2023-02-14:
@gmalinve the branch is based on d1830e889934269ff70b0c043b2853ada3b4b84c which is the commit from main just prior to the current one committed a few minutes ago.  These two tests are failing for me in 2022 R2 at least even with the latest main, however:

```
(venv) PS C:\git\Ansys\pyaedt> pytest _unittest/test_21_Circuit.py::TestClass::test_06b_add_3dlayout_component _unittest/test_21_Circuit.py::TestClass::test_07a_push_excitation
=================================================================================== test session starts ====================================================================================
platform win32 -- Python 3.9.12, pytest-7.2.1, pluggy-1.0.0
rootdir: C:\git\Ansys\pyaedt, configfile: pytest.ini
plugins: cov-4.0.0, xdist-3.2.0
collected 2 items

_unittest\test_21_Circuit.py FF                                                                                                                                                       [100%]

========================================================================================= FAILURES ========================================================================================= 
________________________________________________________________________ TestClass.test_06b_add_3dlayout_component _________________________________________________________________________ 

self = <_unittest.test_21_Circuit.TestClass object at 0x0000029CA95BFAC0>

    def test_06b_add_3dlayout_component(self):
        myedb = self.aedtapp.modeler.schematic.add_subcircuit_3dlayout("Galileo_G87173_204")
        assert type(myedb.id) is int
        ports = myedb.pins
        tx = ports
        rx = ports
        insertions = ["dB(S({},{}))".format(i.name, j.name) for i, j in zip(tx, rx)]
>       assert self.aedtapp.post.create_report(
            insertions,
            self.aedtapp.nominal_adaptive,
            plotname="Insertion Losses",
            plot_type="Rectangular Plot",
            report_category="Standard",
            subdesign_id=myedb.id,
        )
E       AssertionError: assert False
E        +  where False = <bound method PostProcessorCommon.create_report of <pyaedt.modules.PostProcessor.CircuitPostProcessor object at 0x0000029CC96A1FA0>>(['dB(S(J3B2.3.USBH2_DP_CH,J3B2.3.USBH2_DP_CH))', 'dB(S(J3B2.2.USBH2_DN_CH,J3B2.2.USBH2_DN_CH))', 'dB(S(J2B3-2,J2B3-2))'], '', plotname='Insertion Losses', plot_type='Rectangular Plot', report_category='Standard', subdesign_id=60)
E        +    where <bound method PostProcessorCommon.create_report of <pyaedt.modules.PostProcessor.CircuitPostProcessor object at 0x0000029CC96A1FA0>> = <pyaedt.modules.PostProcessor.CircuitPostProcessor object at 0x0000029CC96A1FA0>.create_report
E        +      where <pyaedt.modules.PostProcessor.CircuitPostProcessor object at 0x0000029CC96A1FA0> = <pyaedt.circuit.Circuit object at 0x0000029CA792CF70>.post
E        +        where <pyaedt.circuit.Circuit object at 0x0000029CA792CF70> = <_unittest.test_21_Circuit.TestClass object at 0x0000029CA95BFAC0>.aedtapp
E        +    and   '' = <pyaedt.circuit.Circuit object at 0x0000029CA792CF70>.nominal_adaptive
E        +      where <pyaedt.circuit.Circuit object at 0x0000029CA792CF70> = <_unittest.test_21_Circuit.TestClass object at 0x0000029CA95BFAC0>.aedtapp
E        +    and   60 = <pyaedt.modeler.circuits.object3dcircuit.CircuitComponent object at 0x0000029CC937C4C0>.id

_unittest\test_21_Circuit.py:107: AssertionError
---------------------------------------------------------------------------------- Captured stdout setup ----------------------------------------------------------------------------------- 
pyaedt INFO: using existing logger.
pyaedt INFO: Launching PyAEDT outside AEDT with CPython and PythonNET.
pyaedt INFO: AEDT installation Path C:\Program Files\AnsysEM\v222\Win64.
pyaedt INFO: Launching AEDT with module PythonNET.
pyaedt INFO: Ansoft.ElectronicsDesktop.2022.2 Started with process ID 28768.
pyaedt INFO: pyaedt v0.7.dev0
pyaedt INFO: Python version 3.9.12 (tags/v3.9.12:b28265d, Mar 23 2022, 23:52:46) [MSC v.1929 64 bit (AMD64)]
pyaedt INFO: AEDT file load (threaded) time: 0.16956424713134766
pyaedt INFO: Project Galileo_t21 has been opened.
pyaedt INFO: No consistent unique design is present. Inserting a new design.
pyaedt INFO: Added design 'Circuit Design_YHA' of type Circuit Design.
pyaedt INFO: Aedt Objects initialized
pyaedt INFO: Variable Manager initialized
pyaedt INFO: Design Loaded
pyaedt INFO: Successfully loaded project materials !
pyaedt INFO: Materials Loaded
pyaedt INFO: AEDT file load (threaded) time: 0.04819369316101074
pyaedt INFO: Project differential_pairs has been opened.
pyaedt INFO: Active Design set to Circuit1
pyaedt INFO: Aedt Objects initialized
pyaedt INFO: Variable Manager initialized
pyaedt INFO: Design Loaded
pyaedt INFO: Successfully loaded project materials !
pyaedt INFO: Materials Loaded
------------------------------------------------------------------------------------ Captured log setup ------------------------------------------------------------------------------------ 
INFO     Global:aedt_logger.py:473 using existing logger.
INFO     Global:aedt_logger.py:473 Launching PyAEDT outside AEDT with CPython and PythonNET.
[... 47 lines cut ...]
pyaedt INFO: Project files removed from handlers.
[... cut ...]
commit 066e2ce7103cab9c405eb4b662ddcb658ccd5f67 (HEAD -> main, origin/main, origin/HEAD)
Author: JDufrenneAnsys <125255495+JDufrenneAnsys@users.noreply.github.com>
Date:   Tue Feb 14 10:10:41 2023 +0000

    2454 : Plot brd outline (#2455)

    * 2454 : Plot brd outline

    - Allows to plot a layer not related to a net
    - If layer name is "Outline" plot it as a contour

    * [pre-commit.ci] auto fixes from pre-commit.com hooks

    for more information, see https://pre-commit.ci

    ---------
```

isaacansys, 2023-02-14:
@gmalinve in addition, these two tests failed during CI in another unrelated PR of mine, where that PR does not touch this file at all.  See https://github.com/pyansys/pyaedt/pull/2441#issuecomment-1423320718

closed 2023-02-14 by maxcapodi78

isaacansys, 2023-02-14:
@gmalinve just to close the loop on this, these two tests in 066e2ce7103cab9c405eb4b662ddcb658ccd5f67 and prior have a hidden dependency on another test that creates a solution setup.

```
pytest _unittest/test_21_Circuit.py::TestClass::test_06b_add_3dlayout_component _unittest/test_21_Circuit.py::TestClass::test_07a_push_excitation
...fails

pytest _unittest/test_21_Circuit.py::TestClass::test_06a_create_setup _unittest/test_21_Circuit.py::TestClass::test_06b_add_3dlayout_component _unittest/test_21_Circuit.py::TestClass::test_07a_push_excitation
...passes
```

PR #2461 (2023-02-14): Fix failing tests
Two tests in test_21_Circuit.py were failing:

- test_06b_add_3dlayout_component appears to expect a linear network analysis setup to exist by way of referencing self.aedtapp.nominal_adaptive but none exists
- test_07a_push_excitations appears to expect a linear network analysis named LNA to exist but none exists

Fixes #2460