ansys/pyaedt #2465: stackup error

louseph, 2023-02-14:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

There is an error when exporting stackup json file

create a fresh 3dlayout design. Import one xml in to stackup. Make dielectric fill as one of system material.

Export stackup json using pyaedt.

It will give an error.

This is because when pyaedt tries to write material information in json, it couldn't fetch the system material properties.

### Steps To Reproduce

create a fresh 3dlayout design. Import one xml in to stackup. Make dielectric fill as one of system material.

Export stackup json using pyaedt.

It will give an error.

This is because when pyaedt tries to write material information in json, it couldn't fetch the system material properties.

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansys-api-meshing-prime==0.1.1
ansys-meshing-prime==0.2.0
appdirs==1.4.4
bcrypt==4.0.1
certifi==2022.12.7
cffi==1.15.1
charset-normalizer==3.0.1
click==8.1.3
clr-loader==0.2.5
colorama==0.4.6
contourpy==1.0.7
cryptography==39.0.0
cycler==0.11.0
dash==2.8.1
dash-access-manager==0.1.3
dash-bootstrap-components==1.3.1
dash-core-components==2.0.0
dash-daq==0.5.0
dash-html-components==2.0.0
dash-table==5.0.0
dnspython==2.3.0
Flask==2.2.2
Flask-Login==0.6.2
fonttools==4.38.0
grpcio==1.51.1
idna==3.4
imageio==2.25.0
itsdangerous==2.1.2
Jinja2==3.1.2
kiwisolver==1.4.4
MarkupSafe==2.1.2
matplotlib==3.6.3
mongoengine==0.26.0
numpy==1.24.1
packaging==23.0
pandas==1.5.3
paramiko==3.0.0
Pillow==9.4.0
plotly==5.13.0
plumbum==1.8.1
pooch==1.6.0
protobuf==3.20.3
psutil==5.9.4
pyaedt==0.6.45
pycparser==2.21
pymongo==4.3.3
PyNaCl==1.5.0
pyparsing==3.0.9
PyQt5==5.15.8
PyQt5-Qt5==5.15.2
PyQt5-sip==12.11.1
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.7.1
pyvista==0.37.0
pywin32==305
requests==2.28.2
rpyc==5.3.0
scooby==0.7.1
six==1.16.0
tenacity==8.2.0
urllib3==1.26.14
vtk==9.2.5
Werkzeug==2.2.2

svandenb-dev, 2023-02-14:
@louseph 
Aedt and EDB are not sharing the material database, that's probably why it is throwing the error. We will work on it.
thanks

svandenb-dev, 2023-02-14:
@louseph 
Quick question, could yo please clarify this sentence ? 
"create a fresh 3dlayout design. Import one xml in to stackup. Make dielectric fill as one of system material.
Export stackup json using pyaedt"
Are the steps prior the export to json with Pyaedt done manually in Aedt ?

ring630, 2023-02-17:
@louseph could you describe your application scenario? I would like to understand why you want to defined stackup in AEDT and then export stackup by pyaedt. pyaedt.edb doesn't have access to AEDT material library.  It is a limitation.

svandenb-dev, 2023-04-18:
closing this issue due to product incompatibility

closed 2023-04-18 by svandenb-dev