ansys/pyaedt #2466: launching 3dlayout from edb in 2021R1 doesn't work

louseph, 2023-02-14:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

in 2021R1, pyaedt.Hfss3dLayout does give error. It failed to execute

### Steps To Reproduce

Use pyaedt.Hfss3dLayout with an edb. it will fail in 2021 R1

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.10

### Installed packages

ansys-api-meshing-prime==0.1.1
ansys-meshing-prime==0.2.0
appdirs==1.4.4
bcrypt==4.0.1
certifi==2022.12.7
cffi==1.15.1
charset-normalizer==3.0.1
click==8.1.3
clr-loader==0.2.5
colorama==0.4.6
contourpy==1.0.7
cryptography==39.0.0
cycler==0.11.0
dash==2.8.1
dash-access-manager==0.1.3
dash-bootstrap-components==1.3.1
dash-core-components==2.0.0
dash-daq==0.5.0
dash-html-components==2.0.0
dash-table==5.0.0
dnspython==2.3.0
Flask==2.2.2
Flask-Login==0.6.2
fonttools==4.38.0
grpcio==1.51.1
idna==3.4
imageio==2.25.0
itsdangerous==2.1.2
Jinja2==3.1.2
kiwisolver==1.4.4
MarkupSafe==2.1.2
matplotlib==3.6.3
mongoengine==0.26.0
numpy==1.24.1
packaging==23.0
pandas==1.5.3
paramiko==3.0.0
Pillow==9.4.0
plotly==5.13.0
plumbum==1.8.1
pooch==1.6.0
protobuf==3.20.3
psutil==5.9.4
pyaedt==0.6.45
pycparser==2.21
pymongo==4.3.3
PyNaCl==1.5.0
pyparsing==3.0.9
PyQt5==5.15.8
PyQt5-Qt5==5.15.2
PyQt5-sip==12.11.1
python-dateutil==2.8.2
pythonnet==3.0.1
pytz==2022.7.1
pyvista==0.37.0
pywin32==305
requests==2.28.2
rpyc==5.3.0
scooby==0.7.1
six==1.16.0
tenacity==8.2.0
urllib3==1.26.14
vtk==9.2.5
Werkzeug==2.2.2

svandenb-dev, 2023-02-14:
@louseph 

could you please share the command line you use to open the edb ? 
Do you have a chance to use a more recent Aedt version ? 
Also please make sure you have the latest Pyaedt package 
pip intall pyaedt --upgrade

thanks

Samuelopez-ansys, 2023-02-14:
PyAEDT requires Ansys Electronics Desktop (AEDT) 2021 R2 or later

https://aedt.docs.pyansys.com/release/0.6/Getting_started/Installation.html

louseph, 2023-02-14:
Hi

One correction. The version is 2022 R1

target_aedb= os.path.join(os.getcwd(),'test.aedb')
h3d = Hfss3dLayout(specified_version="2022.1", projectname=target_aedb, non_graphical=non_graphical)

Thanks & Regards,
Lindo Ouseph / Application Engineer

[cid:4818cba2-3d37-4408-92ec-65b40aac4853]

Bangalore / India

***@***.***  /  www.ansys.com<http://www.ansys.com/>

[External Sender]

svandenb-dev, 2023-02-16:
@louseph 
could you please add your error log here ? I am not able to reproduce your issue.
Thanks

svandenb-dev, 2023-02-23:
@louseph one thing to check would be if aedt file already exists and maybe if the EDB were saved in an later version (not sure it would a pb but worth to check)

svandenb-dev, 2023-03-16:
Hi @louseph did you checked this issue ? wanted to know if this can be closed

maxcapodi78, 2023-03-16:
Hi @louseph  AEDT Fails to import an EDB if the project with a same name exists in the same folder. (Example project.aedb and in the same folder there is project.aedt). this behaviour is reproducible in UI where you would get an error of project existing. 
One approach to use could be that if I a project.aedt exists I open that one instead of trying to reimport the edb file

louseph, 2023-03-20:
The issue is happening only with 2022 R1.
For 2023 R1, it was working fine

Thanks & Regards,
Lindo Ouseph / Application Engineer

[cid:cc948368-f5a6-487b-a964-8e5c6418e7c0]

Bangalore / India

***@***.***  /  www.ansys.com<http://www.ansys.com/>

[External Sender]

Samuelopez-ansys, 2023-04-17:
@louseph Please could you check if the issue is solved and close the issue?

closed 2023-06-29 by louseph