ansys/pyaedt #2508: Primitives3DLayout.place_3d_component() does not allow z-displacement

dcrawforAtAnsys, 2023-02-21:
### Description of the feature

We need to implement z-displacement when placing 3D components to place components without intersecting existing geometry.

### Steps for implementing the feature

Native API Syntax to set z-position. Needs to be added to the method.
```
oProject = oDesktop.GetActiveProject()
oDesign = oProject.GetActiveDesign()
oEditor = oDesign.SetActiveEditor("Layout")
oEditor.ChangeProperty(
	[
		"NAME:AllTabs",
		[
			"NAME:BaseElementTab",
			[
				"NAME:PropServers", 
				"92"
			],
			[
				"NAME:ChangedProps",
				[
					"NAME:Location",
					"X:="			, "-0.8mm",
					"Y:="			, "0mm",
					"Z:="			, "0.6285mm"
				]
			]
		]
	])
```

svandenb-dev, 2023-02-23:
@dcrawforAtAnsys 

You can move the 3D component assuming this code, placing a component then moving and rotating.
Is it something you are looking ?

  cmp_instance = hfss.modeler.insert_3d_component(cmp_file, name=cmp_refdes)
  if cmp_instance:
        hfss.modeler.move(cmp_refdes, [1, 1, 1])
        hfss.modeler.rotate(90)

maxcapodi78, 2023-09-27:
@dcrawforAtAnsys  I've fixed it. you can change location of a placed component also with setter location:
com.location = [1,2,3]

PR #3654 (2023-09-27): Added possibility to change the z location of a 3d component when placed in 3d Layout
Fix #2508
closed 2023-09-27 by maxcapodi78