ansys/pyaedt #2514: Bug located in hfss3dlayout.modeler.place_3d_component(…)

gene820715, 2023-02-22:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I'm using v0.6.47 PyAEDT with Python 3.8.6 in Linux CentOS 7 environment. I'm getting an error below.

```
h.modeler.create_3dcomponent('/create_blocks/testing.a3dcomp', component_name='refdes_name', component_outline="None", replace_names=True, hide_contents=True)

File "/lib/python3.8/site-packages/pyaedt/modeler/modeler3d.py", line 216, in create_3dcomponent
    if "CreateRegion:1" in self.oeditor.GetChildObject(el).GetChildNames():
RuntimeError: Error 0x80020009;  Logs or message window might have more details
pyaedt ERROR: Error in :   File "/lib/python3.8/site-packages/pyaedt/modeler/modeler3d.py", line 216, in create_3dcomponent
pyaedt ERROR:  - create_3dcomponent -  GENERAL OR AEDT ERROR.
pyaedt ERROR: Project: CM03_gene (3D Component), Design: CM03 (Modal Network), [error] Script macro error: RadiatingSurface2 is not a child name of 3D Modeler. (10:29:46 PM  Feb 21, 2023)
```

However, I do not have an object called RadiatingSurface2.
```
h.modeler.solid_bodies
['dielectric', 'planting', 'terminal', 'electrodeA', 'RadiatingSurface2', 'electrodeA_1']
```
[image]

### Steps To Reproduce

- Open a .a3dmodel file with HFSS
- Make some modifications such as changing coordinate system
- Create a new 3d model with hfss3dlayout.modeler.place_3d_component()

### Which Operating System are you using?

Linux

### Which Python version are you using?

3.8

### Installed packages

pyaedt==0.6.47 and other dependent packages

Samuelopez-ansys, 2023-03-02:
Hi @gene820715 ,

We debug your issue and we identified a problem in AEDT when a 3DComponent is edited. We figure out how to overcome this limitation and it has been fixed, please check it:
#2560 

The solution for your problem is then:
# Import the 3DComponent
h2 = Hfss(specified_version=ansys_version, non_graphical=False, designname="new", projectname="newProject")
comp = h2.modeler.insert_3d_component(comp_file=os.path.join(cwd, 'test.a3dcomp'))

# Flatten 3DComponent
h2.flatten_3d_components(purge_history=False)

# Modify design
cs = h2.modeler.create_coordinate_system(["-0.2 mm", "-0.15 mm", "0.15mm"])

# Create the new 3DComponent with your modifications
h2.modeler.create_3dcomponent(os.path.join(cwd, 'new.a3dcomp'), component_name='refdes_apn', reference_cs=cs.name,
                             component_outline="None", replace_names=True, hide_contents=True)

Please close this issue if it works.

Regards,
Samuel

gene820715, 2023-03-17:
Works with newest update of pyaedt (v0.6.56). I had to save the project before creating 3d component.

closed 2023-03-17 by gene820715