ansys/pyaedt #2528: Bug located in ...module 'pyaedt' has no attribute 'downloads'

kunal-3196, 2023-02-24:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

module 'pyaedt' has no attribute 'downloads'

### Steps To Reproduce

after importing '*' from pyaedt, same issue is occuring

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.9

### Installed packages

module 'pyaedt' has no attribute 'downloads'

not sure if bug or not

gmalinve, 2023-02-27:
Hi @kunal-3196 ,

Have you tried to import as: from pyaedt.downloads import * if you want to import all methods or otherwise import only the method you are interested, for example: from pyaedt.downloads import download_file ?

Kind regards,

Giulia

kunal-3196, 2023-03-04:
[image]
its not having downloads

gmalinve, 2023-03-06:
Hi @kunal-3196 ,

Which version are you using? could you please download the latest release and see if the issue persist?

Thanks,

Giulia

closed 2023-03-23 by Samuelopez-ansys