ansys/pyaedt #2531: Bug Python exits with many error messages when trying to use Hfss()

jpotter0, 2023-02-25:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

My Python program using pyaedt was working three weeks ago. Now it crashes with multiple error messages on a call to Hfss, with or without arguments, either aedtapp = Hfss(projectname=project_name, designname=design_name, solution_type="Eigenmode",specified_version=version) or aedtapp = Hfss() causes program to crash.

### Steps To Reproduce

"""Script for IHDTL HFSS design using PyAEDT."""

# %% Imports
import math
import os
from typing import TextIO

import pyaedt
from pyaedt import Desktop
from pyaedt import Hfss

# %% Code

directory = "C:\Jims Files\\Nusano\IHDTL"
tanknumb = "7"
tank_name = 'C:\Jims Files\\Nusano\IHDTL\IHDTL_Tank'+tanknumb+'.aedt'
project_name = tank_name

design_name = "ihdtl"
version = "2022.2"
aedtapp = Hfss(projectname=project_name, designname=design_name, solution_type="Eigenmode",specified_version=version)
#aedtapp = Hfss()

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

[image]

Samuelopez-ansys, 2023-02-27:
@jpotter0 Please install a new virtual environment with only PyAEDT and test it. If does work, install Pyside2 and test it again. Let us know what happens.

jpotter0, 2023-02-27:
OK, Samuel. I will do that.

jpotter0, 2023-03-03:
Creatin a  new environment fixed the problem. I was unable to reproduce the problem in the new environment.

closed 2023-03-03 by jpotter0