ansys/pyaedt #2533: I created a new environment for pyaedt and switched to it in Pycharm.

jpotter0, 2023-02-27:
Now I get an error message "Cannot import Desktop from pyaedt". What is the fix?

Samuelopez-ansys, 2023-02-28:
@jpotter0  please give more info. How did you install the environment?
Could you share the installed libraries? pip freeze

jpotter0, 2023-02-28:
> <https://github.com/pyansys/pyaedt/issues/2533#issuecomment-1448203069>view
> it on GitHub, or

Samuel,

I created a new environment. That stopped the 
program from crashing. However, it still doesn't 
find version 2023.1. I set 
pyaedt.settings.use_grpc_api = True. It runs with version 2022.2.

I haven't tried putting PySide2 into the 
environment. I've been busy adding a couple of 
new features and debugging the code. Still having 
a minor issue. When I get it figured out I will 
add PySide2 to the new environment and see what happens.
I'll pass the information on when I get it. I 
still don't know why I can't access version 2023.1

Best regards,

Jim

James M. Potter, PhD, President
JP Accelerator Works, Inc.
2245 47th Street
Los Alamos, NM 87544

TEL: 505-690-8701

jpotter0, 2023-03-01:
I made a lot of progress. Now I am able to use version 2023.1.

I tried to pip install PSide2 in the new 
environment, but it seem to access only the old environment.

Samuelopez-ansys, 2023-03-01:
You could try to install Pyside2 using Pycharm tools: Settings>Project interpreter and install from there.

Please let us know if we could proceed to close the issues.

jpotter0, 2023-03-03:
If I don't specify the version it opens version 2023.1. If I specify the version as 2023.1 it gives an error message, Can't find the file. If I specify version 2022.2 it opens with grpc.

Samuelopez-ansys, 2023-03-03:
Hi @jpotter0 ,

The version should be passed as a string: "2023.1". 

About the grpc, if you want to disable it, before opening AEDT:

pyaedt.settings.use_gprc.use_grpc_api = False

It is strange because in 2022.R2, is not set by default GRCP. But in 2023R1, you need to disable it if you don't want to use GRPC.

jpotter0, 2023-03-03:
Thanks, Samuel!

closed 2023-03-03 by jpotter0