ansys/pyaedt #2551: Potential incompatibility with the rest of PyAnsys libraries

RobPasMue, 2023-03-01:
On the https://github.com/pyansys/pyansys-dev repository (in which we check that all "dev" versions for the public PyAnsys projects are compatible) we have identified some issues with PyAEDT's compatibility with the rest of the libraries. It seems like you are pinning down a specific Python version (3.7). Though support is expected to be dropped for 3.7.

https://github.com/pyansys/pyansys-dev/actions/runs/4299440263/jobs/7494650056

It seems that PyAEDT is forcing 1.24.2 for Python3.7, which is not possible since that numpy version is not supported by Python 3.7. You should adapt your setup.py file to allow lower numpy versions for Pytthon 3.7. Probably ``numpy==1.21.6``.

Pinging @MaxJPRey

maxcapodi78, 2023-03-01:
@RobPasMue thanks for pointing this.

RobPasMue, 2023-03-02:
Problem was with the way that the system version was being parsed. This has been solved by @maxcapodi78 in #2546 =)

Thanks @maxcapodi78!

closed 2023-03-02 by RobPasMue