ansys/pyaedt #2565: How do I create Local Variables for HFSS in pyaedt?

jpotter0, 2023-03-03:
I have some geometric parameters, which I read in from a CSV file. I need to create local variables for some of these parameters so that I can adjust them after I create the model. I used to know how to do it, but apparently I have forgotten. The technique to create an initial value is aedtapp["var"] = "1mm". or some other numerical value. I need to replace that value with an initial value I read in from the CSV file. Then I need to be able to change it and have the geometry change. I use the variables in creating my geometry, but the numerical values are what shows up in the model. Please remind me of how to do this.

jpotter0, 2023-03-03:
I am still struggling with parameterizing my HFSS model. I can get the parameters to be listed in the Design Properties window. However, changing the value in the window has no effect on the value of the parameter. Also, the parameter does not show in the design, just the numeric values of the parameters. How do I create a parameterized model using pyaedt?

jpotter0, 2023-03-04:
I figured it out by going back to a program I wrote in iPython before there was pyaedt. I remembered that the variables had to have two names, but I forgot the details. When i went back to the old code I was able to use it with the pyaedt definition of oDesign. Then I went to ChatGPT. It has some issues with inventing methods or using methods that are in old versions. It keeps trying by making the code more complex. It's like a student fresh out of school with a lot of book learning, but no experience. I had to teach it how to draw a box in pyaedt,

I got the old HFSS Python code modified for the box. Then I asked ChatGPT to translate it to run in pyaedt. It did the right thing, but it had some strange way of formulating the values for the local variables.

Here is a sample that works. I used EDT 2023.1 and pyaed v. 0.6.48.

import pyaedt
from pyaedt import Hfss

# Connect to an active HFSS design session
hfss = Hfss()
# Define the initial values
_x0 = 0.0
_y0 = 0.0
_z0 = 0.0
_x_size = 10.0
_y_size = 10.0
_z_size = 10.0

# Define the parameters for the box dimensions
hfss["x0"] = _x0
hfss["y0"] = _y0
hfss["z0"] = _z0
hfss["x_size"] = _x_size
hfss["y_size"] = _y_size
hfss["z_size"] = _z_size

# Create the box with the parameterized dimensions
box = hfss.modeler.primitives.create_box(["x0", "y0", "z0"], ["x_size", "y_size", "z_size"], "MyBox")

jpotter0, 2023-03-06:
The above code creates a box just fine. The text names are in the box definition, and when I change the values in the Design Properties table and click Apply or OK the box updates to the new dimension. I seem unable to do the same thing with create_polyline. I tried primitives.create_polyline with no change. Is there some way to create a polyline with the quoted names in the polyline definition? I have done this manually for years with no problem, and I did it in an iPython Script before there was a pyaedt.

Here's the line of code defining the points and the crete_polyline. All the values are with the associated names in the Design Properties table.

    LEE_cell_points = [[0.0, 0.0,0.0],[0.0, "rLEE",0.0], [0.0, "rLEE", "lenLEE"],[0.0, "R1", "lenLEE"],
                       [0.0,"R1", "z1"],[0.0,"Lstem","z1"],[0.0,"Lstem", "Lhcell1"],
                       [0.0,0.0,"Lhcell1"],[.0,0.0,0.0]]
    aedtapp.modeler.primitives.primitives.create_polyline(LEE_cell_points,cover_surface=True, name="LEE_cell")

Samuelopez-ansys, 2023-03-06:
Hi @jpotter0 ,

Please test this code:

```
import pyaedt

app = pyaedt.Hfss(specified_version="2022.2")

app["a"] = "2mm"

app.modeler.create_polyline(position_list=[[0.0, 0.0, 0.0], [0.0, "a", 0.0]])

app.release_desktop()
```

It works for me. Please add a full snippet code to reproduce the issue.

Thank you

jpotter0, 2023-03-06:
Samuel, Thanks! I think the problem is that the documentation for create_polyline shows the position list as a positional argument, not a named one. This was compounded by an issue with units. I had some mm parameters that were being treated as meters. It's an HFSS issue I've run into before. Even though my default units are mm, any number entered into Design Properties is assumed to be in meters. I used change_property to define some variables as mm, but I omitted two. Following your example, I used an assigned value position_list =.   Here's the line of code:  aedtapp.modeler.create_polyline(position_list=LEE_cell_points,cover_surface=True,name="LEE_cell").

It's frustrating when the documentation isn't clear.

jpotter0, 2023-03-07:
It seems that I can use a positional argument in create_polyline if the data evaluates to a number, but not if it's a string as needed to use Design Properties parameters. The problem is now solved.

closed 2023-03-07 by jpotter0