ansys/pyaedt #2566: I have a problem trying to plot HFFS data as a funtion of position along a line

jpotter0, 2023-03-03:
I have a parameter Ezz which is defined as the Z component of the Electric Field. I have a line with the name "axis". I want to plot Ezz along the line. Here's what I have: 

aedtapp.post.create_report('Ezz',primary_sweep_variable='Distance',plot_type='Rectangular Plot',
    context='axis',plotname='Ez(z) on axis',polyline_points=1001)

It creates a graph in an HFSS window. The axes are labeled Y (vertical) and X (horizontal). Both have a scale from 0 to 1.0 and no data is displayed. If I create a Fields plot, rectangular and select the path "axis" Ezz in HFSS I get the plot I want. I'm trying to create it in a pyaedt program.

jpotter0, 2023-03-06:
When I can't make headway with pyaedt, I adapt the copied HFSS script to pyaedt using the odesign and oeditor methods. I'm calling it a night now that I've  conquered the create_polyline problem. I continue to struggle with the documentation. In addition to finding that the input list for create_polyline is not a positional parameter, struggle to understand which methods   use the method I need. Is there somewhere that shows the tree of dependencies? Again, I find the doc"

Samuelopez-ansys, 2023-04-11:
@jpotter0 Please could you share a self-contained project to reproduce the issue?
The script should:

1. Create HFSS design.
2. Create geometry
3. Analyze
4. Reproduce the issue you mentioned
5. 
Thanks

jpotter0, 2023-04-12:
I think I got this working, but I'm just getting back to the coed after working on two other projects. I am planning to verify this, but a change to my directory structure is causing error messages. I'm trying to sort it out.

Here is what works:

aedtapp.post.create_report("Ezz", setup_name="Setup", sweep_name="Sweep", solution_type="Sweep",
                            solution_name="HFSS Setup1 : LastAdaptive", report_type="Fields",
                            report_name="Ez(z) on axis", report_category="Create Fields Report",
                            plot_type="Line Plot", context="Axis",
                            expression="E Field/Ez", legend="Ez(z)",
                            x_range_name="Full", y_range_name="Full", z_range_name="Full",
                            polyline_points=1001, use_existing_surf_report=False)

I'll get back to you after I fix my current problem.

Jim

jpotter0, 2023-04-12:
I remembered wrong. It doesn't work yet. I get an error message: 

pyaedt ERROR: create_report() got an unexpected keyword argument 'setup_name' - create_report -  TYPE ERROR.

I'm not up to creating a simple case tonight. I'll try to get something tomorrow.

Jim

jpotter0, 2023-04-13:
Here's a short code that reproduces the problem:

[E-field plot.txt](https://github.com/pyansys/pyaedt/files/11218389/E-field.plot.txt)

Samuelopez-ansys, 2023-04-13:
Hi @jpotter0 ,

Please try the following code, it is working for me:

```
"""Script for Testing create-report in PyAEDT."""
#import math
import pyaedt
from pyaedt import Hfss

directory = r"C:\Users\slopez\Downloads"
tank_name = directory+'/TEST.aedt'
project_name = tank_name
design_name = "test"
aedtapp = Hfss(projectname=project_name, designname=design_name, solution_type="Eigenmode", specified_version="2023.1")

# Geometry
cyl = aedtapp.modeler.create_cylinder(cs_axis="Z", position=[0.0, 0.0, ], radius=100.0, height=10.0, name="cell1",
                                    matname="vacuum")
axis_name = "axis"
line = aedtapp.modeler.create_polyline([[0.0, 0.0, 0.0], [0.0, 0.0, 10.0]], segment_type=["Line"],
                                    name=axis_name,non_model=False)

# Mesh
initial_mesh = aedtapp.mesh.initial_mesh_settings
initial_mesh.auto_update = False
initial_mesh.props["GlobalSurfApproximation"]["SliderMeshSettings"] = 6
initial_mesh.props["GlobalCurvilinear"]["Apply"] = True
initial_mesh.props["DynamicSurfaceResolution"] = True
initial_mesh.update()

#Setup
setup = aedtapp.create_setup(setupname="Setup")

setup.props["MinimumFrequency"] = "1GHz"
setup.props["NumModes"] = 1
setup.props["MaxDeltaFreq"] = 0.1
setup.props["ConvergeOnRealFreq"] = True
setup.props["MaximumPasses"] = 2
setup.props["MinimumPasses"] = 1
setup.props["MinimumConvergedPasses"] = 1
setup.props["PercentRefinement"] = 30
setup.props["MinimumConvergedPasses"] = 1
setup.props["MinimumConvergedPasses"] = 1

# Run
running = aedtapp.analyze_setup(setup.name)

# Edit source
energy = "1"    #Stored Energy in Joules
oModule = aedtapp.odesign.GetModule("Solutions")
oModule.EditSources(
    [
        [
            "FieldType:="	, "EigenStoredEnergy"
        ],
        [
            "Name:="		, "Modes",
            "Magnitudes:="		, [energy]
        ]
    ])

available_variations = {"Distance": ["All"], "Phase": ["0deg"]}
aedtapp.post.create_report(
    "Mag_E",setup_sweep_name="Setup : LastAdaptive",domain='Sweep',primary_sweep_variable='Distance',
    report_category="Fields",plot_type='Rectangular Plot',context='axis',plotname='Ez(z) on axis',
    polyline_points=1001, variations=available_variations,
    )

aedtapp.save_project()
aedtapp.release_desktop()
print("End of script")`
```

jpotter0, 2023-04-13:
Thanks, you also answer my issues with setup and mesh.

However, the plot I get is scaled 0-1 V/m and 0-1mm.
[image]
 ALl I did was copy your code verbatim and changed the name of the directory.

[E-field plot-Samuel.txt](https://github.com/pyansys/pyaedt/files/11220501/E-field.plot-Samuel.txt)

Why don't I get the data plotted?

Jim

Samuelopez-ansys, 2023-04-13:
Hi @jpotter0 ,

Please check if the setup is solved. In my case:
[image]

jpotter0, 2023-04-13:
The issue must be the phase. The code works fine with ComplexMag_E. My plot will be a calculated data variable, the z component of the magnitude of the complex E field.

Thanks again for your assistance.

Jim

jpotter0, 2023-04-13:
Yes that's the same plot I'm getting for the ComplexMag_E

closed 2023-04-13 by Samuelopez-ansys