ansys/pyaedt #2611: python get stuck after error

cristianbabetto, 2023-03-10:
### Before submitting the issue

- [X] I have searched among the existing issues
- [X] I am using a Python virtual environment

### Description of the bug

I hope this is the right place to rise a problem,

I'm using pyaedt with Maxwell,
When I execute a wrong string in python the console gets stuck and I need to kill the Maxwell process.

[image]

### Steps To Reproduce

For example,

m3d = Maxwell3d(projectname='test', designname='test', non_graphical=False, new_desktop_session=True)
m3d['sim_I_peak'] = 'pwl()'

### Which Operating System are you using?

Windows

### Which Python version are you using?

3.8

### Installed packages

standard

maxcapodi78, 2023-03-16:
@cristianbabetto Hello, the problem is that once you connect to an aedt session you need to release it correctly otherwise it won't allow it. This is a limitation of the AEDT API.
Few workarounds:
- use release_desktop method at the end. if the method fails you can still reconnect to that session (maybe with another call to Maxwell3d and then release it
- use the with. Example
with Desktop() as d:
    m3d = Maxwell3d(projectname='test', designname='test', non_graphical=False, new_desktop_session=True)
    m3d['sim_I_peak'] = 'pwl()'

in this way any error in the with will trigger the exit (which will trigger the release_desktop).

I hope it helps
Massimo

cristianbabetto, 2023-03-16:
@maxcapodi78 I tried 23R1 with the latest pyaedt 0.6.55 on one of my PC and I got this error, but I have never noticed this issue. In fact, on my second PC with 21R2 and pyaedt 0.6.22 I get an error but the console is still responding so that I can enter a correct command or execute further commands. I uninstalled 23R1 for space reasons so now I can't try again but I'm quite sure that this strange behavior was not present in the past. I worked a bit with pyaedt and I remember I was able to "try commands and retry again" without this problem. 
Please, let me know.

cristianbabetto, 2023-03-20:
@maxcapodi78 Could you please confirm or not that in the previous combination of Maxwell 21R2 and pyaedt 0.6.22 there were no issues like this?

Samuelopez-ansys, 2023-03-23:
Hi @cristianbabetto ,

We can not reproduce this issue, I think as @maxcapodi78 mentioned, the AEDT session was not correctly released. Please try again, in version 23.1, PyAEDT is connecting AEDT through GRPC, you can disable it and use the legacy COM:

import pyaedt

pyaedt.settings.use_grpc_api = False

app = pyaedt.Hfss(specified_version="2022.2", non_graphical=False, new_desktop_session=True)

app.release_desktop()

Please close the issue if you think it could be closed.

cristianbabetto, 2023-03-23:
Thanks for tour support.

closed 2023-03-23 by cristianbabetto